Jyers/Marlin #937: Cannot resume print or stuck in reheat when pause print

catssont, 2021-05-31:
I updated to the latest release, 2 issues I noticed is when I paused print to change filament,  it keeps showing me the reheat screen and not able to go back to resume print. 

Then I tried to turn off the printer, usually it will prompt if I want to resume print but now it's simply gone and I need to start print all over.

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

HermanGaviriaMarin, 2021-05-31:
Yea I have experienced the same issue

gothcha, 2021-05-31:
@Jyers I can too confirm this bug. 1.3.3 self compile UBL. When print paused it moves to park and nozzle & bed stay at set temp, after around 30 sec it beeps three times and displays heating timed out & prompts for continue. Hot end starts to cool, bed stays at temp. After pressing Continue Hotend heats to temp but printing does not continue, after another 30 secs ‘heating timed out’ is shown with prompt to Continue and process repeats itself ….

gothcha, 2021-05-31:
More info, on 1.3.2 (last before move) the response is different. After Pause but before any timeout, print is continued, it displays the purge more prompt, if cancelled it then continues printing fine but returns to main menu with prepare highlighted and no way to stop or pause print.

if Pause and wait for timeout this works as expected and when reheat selected it displays reheating in the status area, when heated requests continue then purge filament, pressing cancel to end process then moves to commence printing, but Print Paused is displayed in status area and screen reverts to main menu and stuck.

gothcha, 2021-05-31:
OK interesting, I flashed an old file I compiled on 2 May (1.3.2 dated 28 April) With this version everything was perfect. After Pause and Print no filament change or purge was displayed and print continued as expected. If heater timeout occurs it reheats and after waiting for responses it moves back to print position displays print paused and I had to press print again to continue.

So I am not certain what the differences are.

tititopher68-dev, 2021-05-31:
@gothcha , 
I think I have resolved this issue. But I can not test it not being at home.
the problem lies in "void CrealityDWINClass::State_Update()".
and more precisely here : if (process == Wait && !paused) Redraw_Menu(true, true);
Indeed, the call of Popup "Heating, please Wait..." change the state of : Process (Process = Wait).
When asked to resume printing, "paused" go to false, the test "if (process == Wait && !paused)" go to true and so we do a Redraw_Menu....But we are not on a menu ! So .... screen frozen on the popup !

It is necessary to modify, in my opinion, the condition "if (process == Wait && !paused) Redraw_Menu(true, true);" in : 
if (process == Wait && !paused) {
      if ((printing) && (active_menu != Tune))) Print_Screen_Icons();
      else Redraw_Menu(true, true);
}
Je pense que le Redraw_Menu concerne uniquement lorsque l'on active Tune durant l'impression...

gothcha, 2021-05-31:
Thanks Christophe, I shall give that a try

tititopher68-dev, 2021-05-31:
@gothcha ,
Edit : Not fixed

gothcha, 2021-05-31:
1.3.3 does not have this section in the function 

void CrealityDWINClass::State_Update() {
  if ((print_job_timer.isRunning() || print_job_timer.isPaused()) != printing) {
    if (!printing) Start_Print(card.isFileOpen());
    else Stop_Print();
  }
  if (print_job_timer.isPaused() != paused) {
    paused = print_job_timer.isPaused();
    if (process == Print) Print_Screen_Icons();
    if (process == Wait && !paused) Redraw_Menu(true, true);
  }
  if (wait_for_user && !(process == Confirm) && !print_job_timer.isPaused()) {
    Confirm_Handler(UserInput);
  }
  #if ENABLED(FILAMENT_RUNOUT_SENSOR)
    static bool ranout = false;
    if (runout.filament_ran_out != ranout) {
      ranout = runout.filament_ran_out;
      if (ranout) Popup_Handler(Runout);
    }
  #endif
}

Cal-LF, 2021-05-31:
Had the same issue a while back. #798

Jyers, 2021-05-31:
I'll get started testing this and see if I can track down the source of the issue.

Jyers, 2021-05-31:
First test was not able to recreate the issue. I started a print , selected change filament from tune, printer parked and unloaded, and prompted to insert filament. I waited until the heater timed out and then selected continue. It reheated and then prompted me to insert filament again. I did this and the rest of the filament change went as expected.

Notes: I was printing from octoprint and I was using my own config which has a 120 second timeout instead of 45.

I'll try with SD and one of the precompiled configs to see if I can recreate the issue.

Cal-LF, 2021-05-31:
I also use 120 seconds for time out. Would this be something to change in the next release?

Jyers, 2021-05-31:
@Cal-LF Have you also not been able to recreate the issue? If the timeout is the issue, there's something else that needs to be fixed, but perhaps the timeout would be good to increase as well.

Cal-LF, 2021-05-31:
I will try a print with these steps.

Jyers, 2021-05-31:
Tried SD on E3V2-UBL-BLTouch-10x10-HS-v4.2.2-v1.3.3, still unable to recreate issue...

gothcha, 2021-05-31:
Test just by pausing from print menu, this causes the problem on my system, either wait for timeout or continue before, both fail for me

gothcha, 2021-05-31:
I have not tested by using filament change from tune, so cannot say this will cause the error

Cal-LF, 2021-05-31:
Pics show sequence. After return to finish print, the screen is at the Tune menu and clicking Back brings it back to the Tune menu. Working as it should.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Jyers, 2021-05-31:
@gothcha Thanks! Now I see the problem and have managed to recreate it. Should be an easy enough fix.

tititopher68-dev, 2021-05-31:
@gothcha @Jyers, This issue is : if we do  PAUSE, we are waiting for  the HEAT TIME OUT et after the popup "HEATING, wait....",  you stay on this popup !

Jyers, 2021-05-31:
@tititopher68-dev Yes, we understand the issue. I already have a fix. Testing it now.

gothcha, 2021-05-31:
Great, I was about to try and film it!

tititopher68-dev, 2021-05-31:
It's already fixed for me

gothcha, 2021-05-31:
Excellent Jacob, I just confirmed that if print is selected before timeout then it continues correctly, I was confusing myself with all the earlier release tests I had done! So hopefully your fix should sort it

Jyers, 2021-05-31:
Done and dusted! Should be fixed with the latest commit. Thanks to everyone who helped out with this :)

closed 2021-05-31 by Jyers

gothcha, 2021-06-01:
Yes now perfect, thank you 😊