Jyers/Marlin #949: V4.2.2 - Mesh Level 5*5 - Power loss recovery not working

Vignesh2659, 2021-06-02:
Hi, in the bin file that I downloaded for the above configuration, the power loss recovery is not working. Request to please look into this. 6 prints wasted because of this. Could you please download the same bin file and flash your Ender 3 V2 to check this

****************************************************************************************************************************************
Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead use one of the following options:

- The Marlin Firmware forum at https://reprap.org/forum/list.php?415
- The MarlinFirmware Facebook Group at https://www.facebook.com/groups/1049718498464482/
- The MarlinFirmware Discord Server at https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the latest "bugfix" branch to see whether the issue is already addressed.

-->

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

gothcha, 2021-06-02:
Yes this is a known issue and is being worked on

Vignesh2659, 2021-06-02:
Oh okay, thanks, will that functionality still work for 3*3 Mesh Level -
4.2.2?

gothcha, 2021-06-02:
Yes when @Jyers releases 1.3.4 it will cover all the existing binaries see #808

closed 2021-06-02 by Jyers

Vignesh2659, 2021-06-02:
Alright, thank you. Will wait.

Thanks,
Vignesh R

Jyers, 2021-06-02:
The marlin dev's didn't add that method, that was almost certainly added by the Creality team. It is exactly that type of special-casing that I am trying to get away from in this firmware. As I said above, if I have to edit M1000.cpp I'll simply add an elif and have it draw the popup. There's no need to overcomplicate it with a flag.

tititopher68-dev, 2021-06-02:
No alas you're wrong, that's exactly what's in the latest Marlin 2.0.8.2 buid ... and if you don't want to take into account the Marlin updates, it's your choice ... .I know that as soon as I offer something, it never suits you ... it's a shame because it will make it easier for you to maintain your firmware ...

tititopher68-dev, 2021-06-02:
And no dev Creality for powerloss.cpp et powerloss.h, it's a Marlin Choice...Sorry....(https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/feature/powerloss.cpp....https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/feature/powerloss.h) even if you don't like the answer ... Just a word of advice, you are young and undoubtedly  talented, but you cannot always be right, you can also be wrong ...

Jyers, 2021-06-02:
@tititopher68-dev I in no way think I am always right. I'm very sorry if it came off that way. I do appreciate your input and ideas, I just feel that at times you have been a little to eager to assume your implementation is the best approach. However this may have made me a bit dismissive lately and for that I apologize.

To the point of the issue at hand. Let me clarify a bit, Not all of the Marlin code was written by the Marlin devs. Especially the DWIN elements. Marlin got the main screen code and the marlin integrations from the Creality team. This is why I said it was by Creality and not Marlin, I was not doubting you that this is the implementation being used in the current marlin build.

The reason this approach is not ideal for this circumstance is because it was built for the original DWIN code. The old DWIN code had no universal way to draw a popup, it was all hard coded into specific instances. This is why they had to use a flag, as it was the only way to communicate to the DWIN code that a popup needed to be drawn. This however is not the case in this firmware. Because this code has a universal popup handler allowing for a popup to be drawn using a functions call, the code can be greatly simplified by utilizing this functionality rather than creating a roundabout method via a flag.

Keeping in line with Marlin is a big part of this project, but this particular element doesn't fall under that guise. The one major change this fork does hope to bring to the main Marlin repo is the removal of special-case DWIN elements strewn throughout the firmware. And this is exactly one of those cases. Ideally I will be able to tie into the MarlinUI aspect of the powerless popup system eventually, but for now a popup handler call from within M1000.cpp may have to do.

Hopefully this makes sense and shows that I did in fact have valid reasoning behind my statements. As always, thank you for your work and for being a part of this community and I'm very sorry if we've gotten off on the wrong foot here.