Jyers/Marlin #954: Reconnecting to Ender 3 V2 with OctoPrint causes firmware crash

tystuyfzand, 2021-06-03:
### Description

When disconnecting/reconnecting from OctoPrint (such as turning the printer off/on again), it shows a "Serial Status Mismatch" when detecting serial connection, which shows an error screen - seemingly due to a firmware crash. If it appears that this is also a OctoPrint issue, I'll open an issue there, but I've only ever seen it happen with this firmware - Unified 2 and stock firmware don't have an issue with this.

The only way to reconnect successfully is to turn off the printer, turn off the OctoPrint server/host (OctoPi in this case), then turn the printer and OctoPrint host back on. Connections will properly work.

### Steps to Reproduce

1. Connect to the printer via OctoPrint
2. Wait ~ 10 seconds
3. Turn the printer off
4. Wait ~ 30 seconds
5. Turn the printer back on
6. Attempt to connect to the printer again

**Expected behavior:** Seamless reconnect to the printer

**Actual behavior:** Serial Detection Error, M112 Printer SHutdown

#### Additional Information

Log from OctoPrint:

```
Changing monitoring state from "Operational" to "Offline after error"
Connection closed, closing down monitor
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 1 port/baudrate candidates: /dev/ttyUSB0@115200
Trying port /dev/ttyUSB0, baudrate 115200
Connecting to port /dev/ttyUSB0, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0x6aa3c810, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Send: N0 M110 N0*125
Recv: //action:prompt_end
Recv: BL24CXX Check succeeded!
Recv: Error:Serial status mismatch
Changing monitoring state from "Detecting serial connection" to "Error"
Send: M112
Send: N1 M112*32
Send: N2 M104 T0 S0*35
Send: N3 M140 S0*102
Changing monitoring state from "Error" to "Offline after error"
Connection closed, closing down monitor
```

Picture of the error on the printer:

[image]

gothcha, 2021-06-03:
See #777

tystuyfzand, 2021-06-03:
I searched but I guess I used the wrong search term, I'll check it out - thanks @gothcha

tystuyfzand, 2021-06-03:
That fixed it, thanks!

closed 2021-06-03 by tystuyfzand