Jyers/Marlin #956: Power loss recovery not working UBL 1.3.3 10x10 mesh BL touch

HermanGaviriaMarin, 2021-06-03:
As the title says, while doing a print late a night the power went out and came back, to my surprise the printer didn't greet me with the resume print prompt. Now the print is a complete failure. Also pause creates a loop where you're unable to resume the print after the heater timer goes out and continues making you reheat forever. Powering off and on also doesn't trigger power loss recovery

gothcha, 2021-06-03:
See #808

HermanGaviriaMarin, 2021-06-03:
It says that it was fixed? Is there a different area where I should go to besides the releases tab to get the latest version that has been fixed or is the version that is fixed the one that Is found in the releases tab? I ask because in the releases tab under the post that says latest it shows a date from a week or two ago (hence before the fixes were released)

gothcha, 2021-06-03:
It is still work in progress as you can see from the discussions. When complete Jyers would be including it in a future release or you can then download & self compile from the code tab.

HermanGaviriaMarin, 2021-06-03:
I noticed he mentioned something about the "commits" I'm assuming that this is where quit patches are released? If so then these quick patches are left for us to compile on our own correct ? However, it looked like it was an issue that was fixed according to what the grandmaster Jyers had said, in your professional opinion is it worth downloading the latest commit as a zip and compiling that version to fix the issue at hand until the official BIN is released?
[image]

HermanGaviriaMarin, 2021-06-03:
NVM, I just noticed Cal-LF responded saying that this did not fix the issue.

gothcha, 2021-06-03:
WIP ... I have not tested it  
In terms of self compiling it depends how much of an issue this is. In the UK power loss is a very rare phenomenon so not an issue for me. I will allow @Jyers to answer on timescale for release

Jyers, 2021-06-03:
I have to do some more testing to see if I can recreate the behavior seen by cal. 

As far as the releases, due to the importance of the recent pause fix the next release will definitely be in a day or two.

Vignesh2659, 2021-06-03:
Oh, so it is not made public yet. I Will track this thread to know when it is done. Thanks

Jyers, 2021-06-04:
Alright, should be fully fixed right now, it will be out in the next release soon.

closed 2021-06-04 by Jyers