Jyers/Marlin #957: Version 1.3.3 - Stepper position status flashing

Sherman-c, 2021-06-03:
Hi
After upgrade to version 1.3.3, I found the stepper motors status bar flashing with '?'
Software version:: 1.3.3
Display firmware: DWIN_SET(Gotcha)

Also Set Home Position under Homing Menu is not working
Steps to replicate the problem
1. Home Z - nozzle moved X=155.0, Y=130.0, Z=11.42
2. Home X - nozzle moved to the edge, X= 0, Y=130.0, Z=11.42
3. Home Y - Bed moved to the edge,  X=0, Y=0, Z=11.42
4. Press Set Home Position - 2 beeps
5. Press Home All
Expect the nozzle will travel to position 0, 0, 11.42
Actual happen - X=155.0, Y=130.0, Z=11.42, same s Home Z
Bug?

Regards
Sherman

HermanGaviriaMarin, 2021-06-03:
Jyers added a blink to the coordinates, this changes once you home all the axis. Idk about the other problems lol, I'm a rookie and thats the only thing I could answer, it ain't much but its honest work.

tititopher68-dev, 2021-06-03:
@Sherman-c It's not an issue but a very small misunderstanding, Set Home Position function is only for people who want to use the printer with a laser head for example. Also for this type of use making a Home All (dedicated to FDM printing) does not make sense ... The Set Home Position is not permanent. It fixes a real position momentarily at 0.0, z. to adapt to the part (to be engraved for example with a laser head). In a Gcode for laser engraving for example, there is no Autohome G28.
If you have a BLTouch (or other leveling probe), it is normal for the HOme to be done in the middle of the bed. This is the HOMING SAFE function (you must in fact avoid having the probe outside the Bed).

Jyers, 2021-06-04:
@tititopher68-dev Well put, that covers everything I believe :)

closed 2021-06-04 by Jyers