Jyers/Marlin #964: Error when compiling: LCD_BED_LEVELING requires a programmable LCD controller

Gil80, 2021-06-05:
### Description
I tried to compile version 1.3.3 as I read that Jyers/Marlin has Bed Leveling working for the LCD of the Ender 3 V2.
I enabled 
MESH_BED_LEVELING,
LCD_BED_LEVELING,
RESTORE_LEVELING_AFTER_G28

The output error:
```
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\tft\tft_fsmc.cpp.o
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1575:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
     #error "LCD_BED_LEVELING requires a programmable LCD controller."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1575:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
     #error "LCD_BED_LEVELING requires a programmable LCD controller."
```

**Expected behavior:**
Since I thought Jyers has Bed Leveling working for the Ender 3, I thought it would compile.

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
[Marlin.zip](https://github.com/Jyers/Marlin/files/6602618/Marlin.zip)

smooreace, 2021-06-05:
Start with one of the pre-built config file sets in \Configuration
Files\E3V2 Templates... then start tweaking. It works perfectly

Gil80, 2021-06-05:
It still fails.
I took the `Marlin-1.3.3\Configuration Files\E3V2 Templates\ManualMesh-3x3` folder and modified the `configuration.h` file.
These are the changes:

```//#define AUTO_BED_LEVELING_BILINEAR
#define MESH_BED_LEVELING

#define MESH_INSET 15 
#define GRID_MAX_POINTS_X 5

#define LCD_BED_LEVELING

#define LEVEL_BED_CORNERS```

The output is the same:

```Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\tft\tft_fsmc.cpp.o
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1575:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
     #error "LCD_BED_LEVELING requires a programmable LCD controller."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1575:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
     #error "LCD_BED_LEVELING requires a programmable LCD controller."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\SPI.cpp:43:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1575:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
     #error "LCD_BED_LEVELING requires a programmable LCD controller."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\MarlinSerial.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1575:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
     #error "LCD_BED_LEVELING requires a programmable LCD controller."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1575:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
     #error "LCD_BED_LEVELING requires a programmable LCD controller."
      ^~~~~
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\tft\tft_spi.cpp.o
Compiling .pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\tft\xpt2046.cpp.o
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_bl24cxx.cpp:29:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1575:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
     #error "LCD_BED_LEVELING requires a programmable LCD controller."
      ^~~~~
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\HAL.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\dogm\../../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\dogm\u8g_com_stm32duino_swspi.cpp:21:
Marlin\src\HAL\STM32F1\dogm\../../../inc/SanityCheck.h:1575:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
     #error "LCD_BED_LEVELING requires a programmable LCD controller."
      ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_flash.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1575:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
     #error "LCD_BED_LEVELING requires a programmable LCD controller."
      ^~~~~
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\HAL_SPI.cpp.o] Error 1
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\SPI.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_if_iic.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:1575:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
     #error "LCD_BED_LEVELING requires a programmable LCD controller."
      ^~~~~
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\MarlinSerial.cpp.o] Error 1
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\Servo.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
[... 24 lines cut ...]
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\eeprom_wired.cpp.o] Error 1
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\onboard_sd.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\tft\../../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\tft\tft_fsmc.cpp:23:
Marlin\src\HAL\STM32F1\tft\../../../inc/SanityCheck.h:1575:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
     #error "LCD_BED_LEVELING requires a programmable LCD controller."
      ^~~~~
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\sdio.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\tft\../../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\tft\tft_spi.cpp:23:
Marlin\src\HAL\STM32F1\tft\../../../inc/SanityCheck.h:1575:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
     #error "LCD_BED_LEVELING requires a programmable LCD controller."
      ^~~~~
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\tft\tft_fsmc.cpp.o] Error 1
In file included from Marlin\src\HAL\STM32F1\tft\../../../inc/MarlinConfig.h:49:0,
[... cut ...]
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\tft\tft_spi.cpp.o] Error 1
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\tft\xpt2046.cpp.o] Error 1
==================================================================================== [FAILED] Took 7.66 seconds ====================================================================================
Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:07.657
```

Cal-LF, 2021-06-05:
https://github.com/MarlinFirmware/Marlin/issues/19202

Jyers, 2021-06-05:
This firmware has LCD leveling enabled by default, enabling it in the configuration file is not supported as marlin has certain restrictions with what kind of LCD allows that define.

MESH_BED_LEVELING is also not supported, use Bilinear with the PROBE_MANUALLY option. For more info take a look at the manual mesh configs templates.

closed 2021-06-05 by Jyers