Jyers/Marlin #965: MESH_BED_LEVELING results in compile error

enderextender, 2021-06-05:
disable BLTOUCH and auto levelling options

Just uncomment MESH_BED_LEVELING 

Marlin\src\lcd\dwin\creality_dwin.cpp:257:35: error: 'z_values' was not declared in this scope
       bed_mesh_t &mesh_z_values = z_values;
                                   ^~~~~~~~
Marlin\src\lcd\dwin\creality_dwin.cpp:257:35: note: suggested alternative: 'Value'
       bed_mesh_t &mesh_z_values = z_values;

enderextender, 2021-06-05:
I took a guess and it seems this is the correct code

bed_mesh_t &mesh_z_values = mbl.z_values;

Jyers, 2021-06-05:
The MESH_BED_LEVELING feature is not supported on this firmware. For manual mesh leveling use Bilinear with the PROBE_MANUALLY option enabled.

For more info you can take a look at the manual mesh templates.

closed 2021-06-05 by Jyers

enderextender, 2021-06-06:
Ok then how about fix this.  doing what you say results in a new error.

```

Marlin\src\lcd\dwin\creality_dwin.cpp: In member function 'void CrealityDWINClass::Menu_Item_Handler(uint8_t, uint8_t, bool)':
Marlin\src\lcd\dwin\creality_dwin.cpp:3827:35: error: 'mesh_z_values' was not declared in this scope
             const float currval = mesh_z_values[mesh_x][mesh_y];
                                   ^~~~~~~~~~~~~
Marlin\src\lcd\dwin\creality_dwin.cpp:3827:35: note: suggested alternative: 'z_values'
             const float currval = mesh_z_values[mesh_x][mesh_y];
                                   ^~~~~~~~~~~~~
                                   z_values
*** [.pio\build\STM32F103RET6_creality\src\src\lcd\dwin\creality_dwin.cpp.o] Error 1
```

Jyers, 2021-06-06:
I just compiled from the latest source using the ManualMesh-3x3 config template with no errors. I would recommend taking a look at those configs and see if you might be missing something. I can assure you the issue is not the source code.

enderextender, 2021-06-07:
Ok so I just used the config that was in the base directory.

 

Maybe some better docs.  But appreciate the effort on the overall project

 

 

Regards,

Mark Williamson

Ender Extender Support

smooreace, 2021-06-07:
The general instruction from the main Marlin repo cover things well