Jyers/Marlin #968: hi, I tried all the last two version for my ender3 v2 and the filament sensor did not work. so, what I missed , please.

muthana11, 2021-06-06:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Jyers, 2021-06-06:
Please try to fill out the form when posting an issue so we know exactly what is going on.

Have you enabled the runout sensor in the advanced menu? Also does it not work at all or does it trigger when filament is in the sensor?

muthana11, 2021-06-06:
It's triggered,  but the printer freezing the when I put filament the printer restart but without resume the project. Sent from my Galaxy

smooreace, 2021-06-06:
You had too many spaces in the 4th sentence

Now more realistically... it shouldn't take much to realize that you asked
a question that is literally impossible to answer because... you gave ZERO
information to work from.

Jyers, 2021-06-06:
@muthana11 Sounds like a hardware issue. Did you rewire your filament sensor?

muthana11, 2021-06-06:
Ok,First I am grateful for your response Second,  I follow this video for hardware installation:https://youtu.be/2l7c3A6_eT4So, U do not know if what he did correct or not.I will appreciate if you have any advice or video!ThanksSent from my Galaxy

Jyers, 2021-06-06:
I believe if it is rebooting that means the outside wire on your connector is switched. So try moving that wire to the other side.

muthana11, 2021-06-06:
Ok, I will do that,  and let you know what I get. Sent from my Galaxy

muthana11, 2021-06-06:
hello,
thanks a lot Jyers for advising, it is work now. As you said it was wiring fault.
I appreciate for your fast response

muthana11, 2021-06-06:
wonderful project, keeping forward

closed 2021-06-06 by muthana11

Jyers, 2021-06-06:
Glad we could get it fixed! Happy printing! :)