Jyers/Marlin #973: Mesh slot defaults to -1

gothcha, 2021-06-07:
Compiled and flashed new 1.3.4 reset defaults and mesh slot defaulted to -1 as we have discussed before.

So I captured a photo this time.

[image]

gothcha, 2021-06-07:
@Jyers maybe this would avoid the problem?        

           Draw_Menu_Item(row, ICON_ReadEEPROM, "Load Mesh");
            }
            else {
                    if (ubl.storage_slot <0) ubl.storage_slot = 0;    
              gcode.process_subcommands_now_P(PSTR("G29 L"));
                 

           Draw_Menu_Item(row, ICON_PrintSize, "Mesh Slot");
                  if (ubl.storage_slot <0) ubl.storage_slot = 0;
                  Draw_Float(ubl.storage_slot, row, false, 1);

tititopher68-dev, 2021-06-07:
@gothcha
As you already mentioned in a discussion, "storage_slot" is a signed variable and set to -1 in ubl.cpp during a Reset. I think it is for Marlin a choice not to affect the Mesh saved in the EEPROM during a Reset Settings. This is not quite an "issue" but it is indeed disturbing and not logical to Marlin that the default slot is not set to 0 after a Reset Settings. What you are proposing seems correct to me. Also add the test when saving (just in case ...).

Jyers, 2021-06-07:
There is a definite reason for this oddity:
`G29 L: Load a mesh. If no index is given, load the previously-activated mesh. The given mesh index will be used for subsequent Load and Store operations.`

The storage_slot variable is used by marlin to save the "previously activated mesh." This why is defaults to -1 on reset as at that point is assumer there is no previously activated mesh. That being said, this implementation definitely doesn't make sense for this firmware's implementation so I will most likely apply something like the fix @gothcha gave.

gothcha, 2021-06-07:
I concur with all you both said, I am sure this could be fixed in the base code unless they are using the -1 to signify that no value has been set, but it still isn’t clear why there is no check or error reported. However no harm in being safe and it does appear to be catching people out.

Jyers, 2021-06-07:
Actually I think I just thought of a fairly elegant way to deal with this that well matches the reason the -1 is there in the first place. I'll probably implement it later this morning, keep an eye out 👀

Jyers, 2021-06-08:
The new mesh slot popup should remove any ambiguity about the slot setting after a reset, while also allowing a quick way to set it back to zero.

closed 2021-06-08 by Jyers

gothcha, 2021-06-08:
Tested this and I managed to force the slot to -1 by flashing/resetting a version (by mistake) using the default configuration files that didn’t enable BLT, then when I flashed with the correct version slot gives -1. Reset to defaults leaves slot -1 . Loading mesh menu brings new popup. I didn’t try to save the mesh with slot at -1.
Great all as expected as it only ever happens once this is a sensible solution.

One minor comment, your new drawn confirm/cancel popup obviously no longer uses my modified icons, I changed the Confirm background colour to match the Cancel as otherwise it always appears as if Confirm is selected. It may be better to use the dark background for both and then the highlight colour indicates the active setting?

tititopher68-dev, 2021-06-08:
@gothcha @Jyers ,there is still a problem when you want to save the Mesh (popup) after Bed Leveling (UBL), if the slot is at -1.  Indeed G29 S with -1 for Marlin:  Print the current Mesh as G-code that can be used to restore the mesh anytime....but no store the Mesh

gothcha, 2021-06-08:
@tititopher68-dev Looking at the code it prompts to set the slot to zero on Save and on Tilt if slot = -1

I didn't really understand what "_if the slot is at -1. Indeed G29 S with -1 for Marlin: Print the current Mesh as G-code that can be used to restore the mesh anytime....but no store the Mesh_ "meant, probably the translation?

tititopher68-dev, 2021-06-08:
@gothcha 
Just look at Popup_Handler(SaveLevel), more precisely: SaveLevel case:
 When we do a Create Mesh in UBL, after acquiring the different points, we have a Popup that asks you to register the Mesh.  And this is where the problem lies.
 There is no slot control in this case.  If this is at -1, the Mesh will not be saved in a slot.
 Also before launching Bed Leveling in UBL (Create New Mesh), we must add the call to the Popup MeshSlot in case ...

tititopher68-dev, 2021-06-08:
@gothcha
 I say this in the event that a user is going to flash his firmware for the first time in UBL and will normally do a reset settings as advised ... Then he will do his Bed leveling and you can be sure that he will not see  before to choose its slot ... (feedback from users who questioned me by asking why the Mesh is not saved even though they have validated Save Mesh just after the end of the Bed probing.

gothcha, 2021-06-08:
Yes I see, @Jyers can add the check into the save mesh function too. Perhaps on the SaveLevel it could indicate the current slot and automatically select 0 if slot is -1.
I think there maybe a confusion as even after all this the general EEPROM settings need to be saved to store the modified slot number.
It would be so much easier if the slot was just set to zero when the Reset was done?

gothcha, 2021-06-08:
In settings.cpp

    // Unified Bed Leveling
    //
    {
      _FIELD_TEST(planner_leveling_active);
      const bool ubl_active = TERN(AUTO_BED_LEVELING_UBL, planner.leveling_active, false);
      const int8_t storage_slot = TERN(AUTO_BED_LEVELING_UBL, ubl.storage_slot, -1);
      EEPROM_WRITE(ubl_active);
      EEPROM_WRITE(storage_slot);
    }

So it actively sets this to -1 if no mesh slot has been written. Not sure why? So perhaps if the slot was set to 0 just before the reset to defaults this would make the problem go away without any of the new code?

Jyers, 2021-06-08:
You're absolutely right @tititopher68-dev I'll fix that as soon as I get home from work

@gothcha While it may seem it would be easier to just have it reset to zero, I actually like the way marlin does this. The main benifit comes when you have multiple meshes. It would be very easy to accidentally forget that flashing or resetting resets what mesh your on and accidentally overwrite the wrong mesh. I think the popup does a good job of both preventing such accidents with multiple meshes and allowing a super easy way to set it zero for those just using one.

grilparto, 2021-06-08:
Looks like you guys have this in hand but I noticed @gothcha made a comment on the change comparison mentioning they were going to try and recreate this behavior.  Think I noticed it when I go to the Mesh Settings and then Zero Current Mesh.  I tried Load Mesh to load up my previous mesh and panicked when it remained at all zeroes.  Looks like the Slot was set to -1.  I assumed it meant "Slot 1" instead of "Slot -1" and thought nothing of it.

gothcha, 2021-06-08:
@Jyers, I agree after more thought, I assume that Marlin use -1 to indicate there is no active mesh, and I can see that users of more than one slot may find the default to zero an issue.

Home from work, I am amazed you have any time that for hehe.

Jyers, 2021-06-08:
> I am amazed you have any time that for hehe

Tell my about it... 😅

tititopher68-dev, 2021-06-08:
@Jyers 
All this is interesting .... We can see new "issues":
Zero current Mesh -> once again the slot is at -1 (you don't see it right away, you have to exit the menu and come back to see it, or do a View Mesh for example ...).
Another point that does not work is the number of slots actually available. The calculation is flawed. At home (using your build for the test), in UBL 10x10, I have the choice between 3 slots (0 to 2), but in reality, slot 0 and 1 is really usable:
For example. I have a mesh in slot 0. I set slot 1 to zero and I save it (which displays .0). So by changing the slot, I have the values ​​of slot 0 or 1 depending on the one selected.
Now I load the mesh of slot 0.
I then choose slot 2 (in theory, the variable is set to 2).
I make a Save mesh. This amounts to copying the data from slot 0 to slot 2.
Slot 2 is still active, I do a View Mesh, the data is displayed.
I now select slot 1. I load it with a Load Mesh. I display it, I have the .0 ...
I select slot 2. I load it with a Load Mesh. I display it and I have ... the .0s of slot 1
I select slot 0. I load it with a Load Mesh. I display it and I have the data for slot 0.
I select slot 2 again. I load it, then display it, and I still have the data for slot 0.
Conclusion, the calculated size (settings.calc_num_meshes ()) is wrong. Bug Marlin certainly. I have done some tests by increasing or decreasing "eeprom_data_size" which is used for the calculation of the storage space available for Slots and I get the same thing ....
(By the way in debugging, I have a message saying that slot 2 is not usable .... also Marlin calculates the maximum theoretical indexes but not the real indexes. Consequently, it keeps the last valid and usable Mesh. ..)

Jyers, 2021-06-08:
@tititopher68-dev Yes that is strange, I will look into the best fix for the zero mesh issue.

As far as the issue of calc_num_meshes(), are you sure it is a marlin bug? It's not very easy for my to look over the code right now to check, but I very well might have just forgotten to subtract one as the meshes are indexed starting at 0 and calc_num_meshes outputs the max number of meshes not the highest index. I would be far quicker to assume I made such a mistake than such a fundamental function in the marlin code be wrong.

gothcha, 2021-06-08:
@Jyers would it make sense to move the slot set menu to just above the Save and Load mesh menu items, rather than the UBL settings menu? This way it would be obvious and easy to define the slot when loading and saving.

Jyers, 2021-06-08:
@gothcha Great point, I think that would be an excellent place to put it.

tititopher68-dev, 2021-06-08:
@Jyers J'ai jeté un oeil sur le code, le nombre de mesh stockable est **settings.calc_num_meshes() -1**

Jyers, 2021-06-08:
@tititopher68-dev Yes, that's what I was saying. Taking a look at my code I had it as:
`Modify_Value(ubl.storage_slot, 0, settings.calc_num_meshes(), 1);`
So it looks like it was just my dumb mistake 😅 

Easy fix at least!

Jyers, 2021-06-09:
Alright... I think I got everything now