Jyers/Marlin #979: baby stepping modifies M851 z-probe offset

dwery, 2021-06-08:
When `#define BABYSTEP_ZPROBE_OFFSET` is **undefined**, babystepping with M290 should **not** modify the z-probe offset. However it seems that when using the display's microstep up/down the M851 value is actually affected, probably because in addition to calling M290  the disply code performs other manipulations.

dwery, 2021-06-08:
Maybe a note on the wiki or an `#error` directive that forces that option to be on would be enough to solve any potential confusion. Personally I prefer to keep the z-offset at the true distance and compensate for printing.

Jyers, 2021-06-08:
That's definitely fair, my ability to improve to docs has been largely inhibited by my lack of time, but I believe I have a large improvement for that coming. Anyway yes, maybe I'll make an error to force that setting off, at it has to be disabled to avoid issues.

Interesting philosophy with the z offset, I feel like that creates more work for you though. Why wouldn't you want to tune your z-offset based on how it prints? Then you wouldn't have to bebystep every time.

No critique so all, I'm just interested in your method :)

closed 2021-06-08 by Jyers

dwery, 2021-06-08:
I'll probably lean toward what you are describing. I built a DIY probe so I'm approaching the whole process one step at a time so that I can both figure it out and attribute issues to - hopefully - one single source. It seemed that I had repeatability issues when changing the offset after the mesh construction. Will try again.

dwery, 2021-06-08:
I've detailed my experiment in this discussion: https://github.com/Jyers/Marlin/discussions/986