Jyers/Marlin #980: Manual mesh bed levelling inconsistency

Gil80, 2021-06-08:
I use manual mesh bed levelling with the Jyers/Marlin firmware.
I initially use manual levelling to set the springs as best as I can. I then engage the manual mesh bed levelling,  but the printer has a Z offset of 0.20 when using the mesh bed levelling.
I have to lower Z till about -0.03 for it to just grab the paper, but at 0.00, the Z carriage clicks the Z stopper.
I don't understand why the Mesh bed levelling is so far off after I've done manual bed levelling. I measure the same point, once using manual levelling, the second time, using the mesh bed levelling. They both sample the same point, but both provide different results... 

**Steps to reproduce:**
1. switch the printer on.
2. Home printer.
3. Heat bed to 60.
4. Go to manual levelling. The nozzle will go to the bottom left corner as a starting point where it's just a few mm from the Z stop switch.
5. level each corner using the wheels so the nozzle is grabbing the paper but I can still move the paper.
6. Repeat the process until I feel the nozzle is the same height on all corners. (I use the yellow springs as part of the recommended upgrade for the printer)
7. Then go to manual mesh bed level.
8. The nozzle will go to the same bottom left corner as before in step 4.
9. I'll then go to macro tune it as presented by the LCD screen of the V2. By default, the Z offset during the macro setting is set to 0.20.
10. It is way too high than my initial manual levelling. To compare it to the same nozzle height as I did in step for, I have to macro step down to -0.03. By this time, the Z stopper is actuated at -0.01.

**Expected result:**
The manual mesh levelling starting point should have the same nozzle height as I calibrated in step 4 for manual bed levelling.

**Actual result:**
When using manual mesh levelling, the nozzle height is too high and require a very large adjustment to match the same result as achieved with manual levelling.

I followed this video by CHEP: https://www.youtube.com/watch?v=9TCqQF374GU&t=250s 
I'm following exactly his steps.

Jyers, 2021-06-08:
I think this might simply be some confusion with how the leveling procedure runs. No matter the point the offset always starts at 0.2, that's just an arbitrary point that's high enough that it should never hit the bed. The point that matters is the -0.03. 0 is the point the manual leveling goes to, so getting an offset of -0.03 means your just 3 hundredths of a millimeter off of your original point. This means you did an excellent job of manual leveling and it's no cause for concern.

Hopefully this makes sense and clears things up a bit.

Gil80, 2021-06-08:
Thank you. That makes more sense now.
Then the last confusing bit is that when I go below 0.0, I hear the Z stop actuating. So if I need to go to -0.03, it's past the actuation point. Isn't this a problem?

Jyers, 2021-06-08:
Actually no, the z limit switch has at least 1mm of safe extra travel below its actuation point before there is any physical obstruction for the axis. So although it may feel strange to go past the trigger point, I assure you the firmware won't let you go too low.

Gil80, 2021-06-08:
thank you. I'll go and redo the mesh now. Appreciate the explanation and the great firmware!

closed 2021-06-08 by Gil80