Jyers/Marlin #981: Filament sensor backwards

ZDITTRIC, 2021-06-08:
### Description

 When using z stop filament sensor it shows filament runout when filament is present

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Jyers, 2021-06-08:
This is due to the two possible pin modes for filament sensors, of which only one can be set for the firmware. It's easy enough to fix by swapping the outside wire on the filament sensor connector to the other side. If you need more info on what exactly to do let me know.

closed 2021-06-08 by Jyers

ZDITTRIC, 2021-06-08:
To confirm, I’m swapping the wires for S (Signal) and V (3.3V) leaving the
ground wire alone? Thanks for the help!

Jyers, 2021-06-08:
That's is correct. The only part I'm a bit fuzzy on is whether or not you should disconnect the Vcc wire. So if you swap the wires and find it not working as expected try disconnecting that wire entirely.

adam3654, 2021-06-08:
> That's is correct. The only part I'm a bit fuzzy on is whether or not you should disconnect the Vcc wire. So if you swap the wires and find it not working as expected try disconnecting that wire entirely.

As far as I'm aware, there is no need for the VCC wire for the sensor to operate correctly. The MCU pin is pulled up internally (#define FIL_RUNOUT_PULLUP) and externally according to the schematic by @gothcha

Jyers, 2021-06-08:
@adam3654 Thanks for the clarification there. Just as a check there should be no negative connotations to connecting the Vcc correct? Just wondering as it is easier to simply say swap the outside wires than to also add that they should disconnect one of them.

gothcha, 2021-06-08:
Yes it is a good idea to remove the Vcc wire to avoid possible shorts. The only benefit to connecting Vcc would be on a sensor with an LED. Only a connection to ground either with the NO or NC switch pins is required.

adam3654, 2021-06-08:
Yeah, what @gothcha said ;-)

I still struggle with the default logic. The sensor could be disconnected from the motherboard and the firmware will still think that the sensor is in a normal state. It would make more sense to tie the MCU pin (PA4?) to ground when filament is present. then the signal would go high if there was a runout situation, or if the sensor became detached.

gothcha, 2021-06-08:
If you compile your own firmware you can specify HIGH or LOW for filament present

Jyers, 2021-06-08:
> the default seems backwards to me

This is the default marlin has set in the configs, it is also the pin mode used by some popular filament sensors such at the btt smart sensor. To be completely honest, I might have switched it a while ago, but at this point I don't want to make a ton of people rewire their sensors again... 😅

ZDITTRIC, 2021-06-08:
> If you compile your own firmware you can specify HIGH or LOW for filament present

This is what I’m thinking, I’m running into errors in VSC (doesn’t like “enabled” any where in the code). If I can work through this issue I think it would work best

adam3654, 2021-06-08:
I have to switch the housing to a shorter one in the near future, so I will re-wire it then and switch it to HIGH, I wanted to keep it the way it was in the configs, because I figured there was a reason, but I have yet to come up with a good reason. I have a vary little bit of controls experience, and I cringe when I think of being able to unplug the runout sensor mid print and the printer would just keep chugging along like nothing is wrong.

Jyers, 2021-06-08:
@ZDITTRIC Check out #932 for a phenomenal fix for all your VSC issues.