Jyers/Marlin #982: Unusual positive offset to mesh when performing Autotilt Current Mesh

grilparto, 2021-06-08:
### Description

When performing "Autotilt Current Mesh" on a newly created active mesh, the resulting mesh pattern appears to add an unexpected positive offset to grid values.

Running [https://github.com/alexqzd/Marlin](alexqzd/Marlin) (fork of Jyers/Marlin), 1.3.4 UBL-BLTouch-15x15-HS.bin on a Voxelab Aquila.

### Steps to Reproduce

Prior to reproducing, the system must have already been manually levelled at four corners.  The system must have the z-offset properly set (mine is -2.67) and saved.  The system must have a 15x15 mesh created and saved.  In **Leveling** menu, **Leveling Active** must be checked.  No changes to the bed leveling (either physically or through the software) prior to trying these steps.  Ideally, the steps should be performed immediately after capturing the 15x15 mesh, and under the same conditions (i.e. bed and nozzle preheated to same temperature).

1. Observe the current mesh by going to **Leveling** > **Mesh Viewer**.
2. In the **Leveling** menu, select and click **Autotilt Current Mesh**.
3. Once the 3-point tilt operation is complete, observe the new mesh within the *Mesh Viewer*.

**Expected behavior:** 

The new mesh should ideally match the old mesh, or appear qualitatively similar with reasonable quantitative variation.

**Actual behavior:**

The new mesh appears to be qualitatively similar to the old mesh, however the new mesh has z-values that have significantly increased.  For example, a mesh with no positive z-values now indicates a mesh with no negative z-values almost as though the new mesh has been offset by the most negative point from the old mesh.

#### Additional Information

Photo of original mesh for example: [image]

Photo of mesh after autotilting: [image]

Some troubleshooting that I performed:

- Tested if Z-Offset Value had any impact.
1.  Set Z-Offset to 0 without saving.  Autotilt produced same results.
2. Set Z-Offset to 0 with saving.  Autotilt produced same results.
3. Set Z-Offset to +5 with saving.  Autotilt produced same results.
4. Tried to issue a G29 J command in OctoPrint, produced same results.
5. Tried to issue a G29 J2 command in OctoPrint, produced similar results with some variation (likely due to 4-points measured).

As a sanity check, I did try a bed leveling pattern that I commonly print (1 layer of 5 squares.  One in each corner, one in the middle).  With the old mesh, the pattern prints very well.  With the autotilted mesh, the pattern prints poorly (nozzle is too high off the board).

I have one workaround in mind, but haven't tried it yet.  I might try adjusting the z-offset value after autotilting using a piece of paper.  I suspect that the Z-offset will decrease from around -2.67 (old mesh) to ~-2.80-2.82, which ought to bring down grid spot 7,7 to Z=0.0 (since it was -0.01 in the old mesh and increased to 0.14 in the new mesh).  Although this workaround wouldn't make using the autotilt feature (G29 J) feasible in the starting gcode for slicer settings.

gothcha, 2021-06-08:
@grilparto Yes this is a little odd, however I may have a possible theory. The 3 point levelling points are chosen automatically, my thought was that if these points do not correspond exactly to a probed mesh point then here will be some interpolation (guess) and this may cause the error you are seeing. In my setup I suspected that this was occurring as the edge points of the 10x10 mesh were not probed and were filled in automatically (by extending mesh) and always appeared a little odd. The auto tilt points used these suspect mesh areas. I changed the mesh and probe margins to ensure all 100 points were able to be probed and then defined the 3 tilt points manually in the configuration file to match the mesh points. Result was that mesh was more consistent over the bed and the tilt always gave the same result as the original mesh.

Sorry this is a little long!

gothcha, 2021-06-08:
You could check this by noting the coordinates of the tilt points and then run the mesh levelling again and see if these 3 points correspond with points that have been probed. If they do then clearly this is not the problem, if they do not then this may indicate where the issue lies

grilparto, 2021-06-09:
@gothcha You're absolutely correct (I did a 15x15 mesh).  I spent the better---er---most--of my "work day" discovering that.  It does look like it picked probing points that aren't necessarily outside the mesh, but are outside the probe-able area. 

I changed my X and Y offsets for the probe to 39.4, and 9.5 respectively, and the bed is 220,220 so that's worth bearing in mind before I explain more.  This is going to be somewhat of a brain spew here; sorry if I meander around.

The 3 points (x,y) picked were:

1. - 49.4, 19.5
2. - 230, 19.5
3. - 139.7, 219.5

Removing the probe offsets that should mean we're probing at:

1. - 10, 10
2. - 190.6, 10
3. - 100.3, 210

So, nah, the chosen points don't align with any of the coordinates on the mesh (I can check the x,y coords for each mesh index in the **Manual Mesh** menu.  For 1, the first mesh grid point 0,0 were (x,y) coordinates (1,1).  Probably because we've set MESH_INSET to 1.  We can't probe it though because PROBING_MARGIN is 10.  I think the first probe-able grid point is 1,1, which corresponds to coordinates (16.5,16.5).

I guess you could try what you've suggested.  Is that setting MESH_INSET to a value that only includes probe-able areas?  It looks like the firmware will try to interpolate points that are "off the mesh" in that case or use UBL_Z_RAISE_WHEN_OFF_MESH if it's defined.  Think it'd need to be set to 30?  That would be assuming you can't move the nozzle any further than x=230, and adding in a PROBING_MARGIN of 10.  (230-40).

The other thought I had would be to hard code the  PROBE_PT_*'s in the Configuration_adv.h such that they align with the exact (x,y) coordinates associated with mesh grid that can (and has been) probed.  Not a great solution though since it ends up being specific to one kind of setup and not as relative as using MESH_INSET.

  //#define PROBE_PT_1_X 15
  //#define PROBE_PT_1_Y 180
  //#define PROBE_PT_2_X 15
  //#define PROBE_PT_2_Y 20
  //#define PROBE_PT_3_X 170
  //#define PROBE_PT_3_Y 20

Guess there's also defining the X and Y min/max for the mesh to try and maximize the mesh area (also in Configuration_adv.h).  That could give us a rectangular mesh where each point is probe-able and none are interpolated I guess.  Not sure if there are problems with that.

```
  // Override the mesh area if the automatic (max) area is too large
  //#define MESH_MIN_X MESH_INSET
  //#define MESH_MIN_Y MESH_INSET
  //#define MESH_MAX_X X_BED_SIZE - (MESH_INSET)
  //#define MESH_MAX_Y Y_BED_SIZE - (MESH_INSET)
```

I'll mess around with it and see if I can figure out how to compile the Marlin firmware tonight (I'm just getting into this 3D printing stuff).  Took me a bit to figure out the nice little bit of scripting that auto-generates the Configuration.h files and whatnot for all the mesh grid-size variations of the Jyers/Alex firmwares, so I should be able to cobble together the settings being used to generate the UBL-BLTouch-15x15-HS.bin firmware and play around with MESH_INSET and whatnot.  

Couple of concerns I'm going to try and look at when I have more time.  Probably the kind of stuff that I suspect an issue with, but will find has been gracefully resolved once I can wrinkle my brain enough to follow the code properly.

- How does the firmware pick the 3 points to check?  --think this is in ubl.h or ubl.cpp
  - I'm assuming based on PROBING_MARGIN and some boundary conditions involving the bed size and whatnot and that it's just trying to pick the outermost points it possibly can taking that all into account.
  - I was surprised it wasn't setup to pick a mesh grid location that 1. Existed and had an associated (x,y) coordinate, and 2. Was possible to probe.  I'd almost prefer to update the code to handle this more robustly instead of adjusting the configuration.h or configuration_adv.h to workaround it...but after a bunch of digging, I still haven't figured out which file has the function (or I just missed it in ubl.cpp or something).

- I understand that the firmware interpolates the unmeasured mesh grid points ... but what offset is it using for x,y coordinates 10,10 which aren't associated with an actual mesh grid for example?
  - Is it interpolating a specific value for those (x,y) coordinates?
  - Or is it going to pull the offset that was interpolated for mesh grid 0,0 (x,y coordinates 1,1)?  <--I think it does this based on what I saw in ubl_motion.cpp but I'm also out of my element digging this deep and this greedily into the code.
  - Or does the firmware incorrectly think it's off the mesh.

- I'm also curious how it handles point number 2 up there.  The point it selected is essentially X_MAX_POS.  I assume it wanted to pick coordinates 210,10 (220 - PROBING MARGIN of 10), and the code recognizes that it would need to move the nozzle beyond X_MAX_POS to achieve it, so it settles for coordinates X_MAX_POS, 10 instead.

grilparto, 2021-06-09:
Alright, well I'm stumped for now.  Long story short, I was able to get things working as I'd expect now.  Creating the mesh and then tilting it right after gave me a result that was really close to the original.  My immediate theory is that it could have something to do with either my process (starting from scratch, manual level, z-offset, etc) or maybe I changed my probe XY offset after capturing the original mesh?  I can't recall now if I'd changed it before or after.  This time around I just left it at 40,9.  Close enough.

I unintentionally changed more than one thing in the firmware than I intended, but here's what I got:

_Configuration.h_
```
was     - #define MESH_INSET 1
is now  + #define MESH_INSET 10
```
```
was     - #define NOZZLE_PARK_Z_RAISE_MIN   5
is now  + #define NOZZLE_PARK_Z_RAISE_MIN   25
```

_Configuration_adv.h_

```
was
- #if EITHER(AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_UBL)
-  //#define PROBE_PT_1_X 15
-  //#define PROBE_PT_1_Y 180
-  //#define PROBE_PT_2_X 15
-  //#define PROBE_PT_2_Y 20
-  //#define PROBE_PT_3_X 170
-  //#define PROBE_PT_3_Y 20
- #endif
```
```
is now
+ #if EITHER(AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_UBL)
+   //#define PROBE_PT_1_X 15
+   //#define PROBE_PT_1_Y 180
+   //#define PROBE_PT_2_X 15
+   //#define PROBE_PT_2_Y 20
+   //#define PROBE_PT_3_X 170
+   //#define PROBE_PT_3_Y 20
+ #endif
```

I'd only intended to make the PROBE_PT changes figuring that I would be probing coordinates that aligned with measurable mesh grid points.  Instead I apparently increased the MESH_INSET by 10 and forgot to undo it (which might've just been the key here).  I wonder if it was intended by @Jyers to be 10 to begin with.  It looks like it's set to 10 in the MESH_BED_LEVELING settings.  It still doesn't allow UBL to probe 100% of the bed but I got 196/225, so I could still go a little further and make the adjustments you've made to the probe margins, but I don't think it'll make a huge difference.  I don't recall how many points I got with the inset set to 1, but much less...I want to say around 100 or 125.

Beyond that I still want to go back and test if things get real wonky if I change the Probe XY offsets and see what happens.  Failing that, I'll start undoing changes.  Something I did caused the issue to begin with.

gothcha, 2021-06-09:
Very interesting, so exactly as I suspected. The tilt points are chosen in the Marlin UBL code so not sure if it will be possible to fix, it is something that would be best to log with the main marlin repo. 

I will send you my info for the config.adv etc with the mesh and probe offsets and how i calculated them and the tilt points. The tilt point is exactly as you postulated, it tries to probe the corners (or edge for 3 points) of the bed then checks the offsets or max travel and uses the limit points, which result in points not necessarily on the mesh. 

It is easy to compile the code, use VS Code, you will have it up and running in no time, there are plenty of tutorials.

gothcha, 2021-06-09:
Oops you beat me to it - just woken up!

OK this is how I did it. My probe offsets are -42, -11
In adv.h
#define MESH_MIN_X 0
#define MESH_MIN_Y 10
#define MESH_MAX_X X_BED_SIZE - 23
#define MESH_MAX_Y Y_BED_SIZE - 15

#define PROBING_MARGIN_LEFT 0
#define PROBING_MARGIN_RIGHT 23
#define PROBING_MARGIN_FRONT 10
#define PROBING_MARGIN_BACK 15

  #define PROBE_PT_1_X 0
  #define PROBE_PT_1_Y 10
  #define PROBE_PT_2_X 207
  #define PROBE_PT_2_Y 10
  #define PROBE_PT_3_X 115
  #define PROBE_PT_3_Y 209

By setting the mesh and margin to the same values it ensures it can probe all the points. The Max_X point is calculated using the max travel-probe offset and Max_Y to allow for the Probe offset with a little margin.

Then the tilt points are the margins of the mesh and the centre point.

Hopefully this will help. The MeshInset in config.h is over-ridden by the mesh_min so you can leave that as it was.

gothcha, 2021-06-09:
Now I look at it PROBE_PT_3 may look odd, I chose a mesh point nearest the centre of the bed for X and a Y point on the mesh near the back. I could have used Y of 215  (MESH_MAX_Y)

The Mesh_Min values need to clear the bed clips.
I have X_MAX_POS 249 but this can be very close to hitting the support so you may want to check this.

Jyers, 2021-06-09:
Just a few notes from me here:

>  I wonder if it was intended by @Jyers to be 10
I don't know exactly what config you're looking at, but all of my UBL templates has MESH_INSET set to 10

>it is something that would be best to log with the main marlin repo
100% agree. If this is an issue with how UBL tilting is handled, that change needs to be made upstream.

If there is anything you need from me here feel free to @ me, or if there's nothing more we can do here we can close this and bring it upstream.

grilparto, 2021-06-09:
> I don't know exactly what config you're looking at, but all of my UBL templates has MESH_INSET set to 10

Found the discrepancy on the MESH_INSET.  Completely missed that I'd grabbed one of the Aquila configuration.h from alexqzd/Marlin.  I'll mention it to him.  It was from [this config file](https://github.com/alexqzd/Marlin/blob/Ender_3_V2_Extensible_UI/Configuration%20Files/Aquila%20Templates/UBL-BLTouch-10x10/Configuration.h), specifically.

Agreed that we can handle this upstream since we're looking a little deeper now at how Marlin handles UBL.  I'll try to gather some more info before submitting an issue there because I'm not entirely sure if I can reliably recreate the issue (though I have some ideas).

I'll close this out.  Thanks for the help!  Feel like I'm sorted out for the most part.

grilparto, 2021-06-09:
Now I recall why I opened this here.  @alexqzd doesn't have an issues tab on their fork, and not being fully understanding of how I ought to go about things on git, I just went to the nearest branch.  Sorry!

Jyers, 2021-06-09:
No problem at all! It's probably best most issues from @alexqzd's fork are brought here unless they are specifically related to the aquilla. Wouldn't want them to get swamped by general firmware issues since it isn't really their job to fix things like that.

tititopher68-dev, 2021-06-09:
@grilparto
The Aquila has a 235mm bed but due to the position of the X-endtop, the usable area is reduced to 220mm.  You can print a new X-axis motor cover on Thingiverse to find the 235mm.  You can also reverse at the level of the Hotend support plate, the screw of the bottom vlsot.  This will allow you to gain another 10mm of margin.  I think @alezqzd did this.  With these modifications you can apply the same settings as on the Ender3V2.