Jyers/Marlin #992: Mesh Viewer is empty after reboot

OberfragGER, 2021-06-11:
### Description

I run the 1.3.4. (E3V2-UBL-BLTouch-10x10-HS-v4.2.2-v1.3.4.bin) release on an ender 3 V2 including BLTouch V3.1.
After a reboot the mesh is not shown.

 When i create a new mesh (10x10), i press on "save to eeprom" and get the answer "saved". The mesh is properly shown on the mesh viewer. With some red and green spots. Everthing is fine.

After a reboot and loading the mesh- nothing is displayed in the mesh! No data, no green and res spots, nothing.

### Steps to Reproduce

1. create new mesh
2. save mesh to eeprom
3. view mesh
4. reboot
5. load mesh

**Expected behavior:** [What you expect to happen]

Mesh is shown

**Actual behavior:** [What actually happens]

No mesh is displayed

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

gothcha, 2021-06-11:
Check the slot setting in UBL Settings, it is likely set to -1. Set to 0 and you should be good to go. Latest commit has the slot setting above the load mesh menu to make this clearer.

OberfragGER, 2021-06-11:
> UBL Settings,

Mh. Sorry. Im pretty new to this stuff. Where are the UBL-Settings? I didnt seen these settings in the printer before (now its printing, no possibilty to look again).

gothcha, 2021-06-11:
Go to Levelling, then Levelling settings, then Mesh Slot and set to zero

OberfragGER, 2021-06-11:
> Sorry - Go to Levelling, then Levelling settings, then Mesh Slot and set to zero

Okay. I found it. No need to level new. After loading mesh- the mesh viewer shows the colors again.

I assume that now im loading every time with the G420 S1 command the "Mesh Level 0"?

gothcha, 2021-06-11:
Yes correct, G420 S1 activates the current mesh. Even safer is G29 L0 this will load the mesh from slot 0 you can follow this with G29 J to run the 3 point tilt if the bed levelling may have moved over time.

OberfragGER, 2021-06-11:
So i have to change the follwoing into Cura (in this oder):

delete: G420 S1
add:
G29 L0
G29 J

right?

gothcha, 2021-06-11:
Exactly

closed 2021-06-11 by Jyers

OberfragGER, 2021-06-11:
Thx a lot!

OberfragGER, 2021-06-11:
THX a lot!