Jyers/Marlin #1000: Cancelling a print via LCD won't cause the nozzle to lift up

Gil80, 2021-06-13:
### Description

If requiring to cancel a print via LCD screen, I'd expect the nozzle to park or at least lift up from the print, otherwise, the hot nozzle is singing the print until it cools down.
Then the print will get stuck to the nozzle when trying to home the nozzle.

**Expected behavior:** get the nozzle up by some Xmm from the print when cancelling the print from the LCD screen.

**Actual behavior:** print stops. motors are disabled. the nozzle is not lifting up. the print gets stuck on the nozzle.

Jyers, 2021-06-13:
Is this when printing from octoprint?

Gil80, 2021-06-14:
either octoprint or printing from SD card.

Jyers, 2021-06-14:
If it's not even functioning after an SD print that indicates something very strange is going on. Moving on an SD cancel is handled by Marlin and I don't do anything that should effect that. Perhaps try clearing eeprom by going to control and selecting reset to default and then store settings. I know it sounds unrelated, but old eeprom can cause some very weird issues.

Gil80, 2021-06-14:
Ok, I'll try that.
Not sure if I should open a new issue or not, but I have a separate question. I'm not sure if this firmware supports Jerk or Junction Deviation.
I think I've seen junction deviation in the code, but not sure.
thank you.

Jyers, 2021-06-14:
At the moment the firmware only supports jerk. You can compile your own version for junction deviation but you would have to set it through a terminal.

darco415, 2021-06-15:
this happens to me as well using Octoprint. I will try the sd ccard and report back

olskar, 2021-06-16:
I have the same issue when using octoprint

cruzer619, 2021-06-16:
for Octoprint only.. you can add gcode to lift the print head and more when you cancel a job..  I go over it in this post : https://github.com/Jyers/Marlin/discussions/366

Gil80, 2021-06-17:
> 
> 
> for Octoprint only.. you can add gcode to lift the print head and more when you cancel a job.. I go over it in this post : #366

Thanks.
I've added the following to my "After print job is cancelled" Gcode:
```M77 ;Stop Print Job on Display
G1 X5 Y5 F3000 ;Wipe out
G1 Z10 ;Raise Z more
;disable all heaters
{% snippet 'disable_hotends' %}
{% snippet 'disable_bed' %}
M106 S0 ;disable fan
G1 X0 Y180 ;Present print
M84 ; disable motors```

Jyers, 2021-06-17:
@Gil80 Yes, canceling from the screen simply sends a command telling octoprint to cancel, so that script will work. 

My real concern here is the failure to lift after canceling from SD. I haven't done an SD print in a while so I'll have to check that out.

cslux, 2021-06-18:
Cancel doesn't work either when printing via Cura. Nothing happens. I'm sure it worked with the manufacturer firmware.

Jyers, 2021-06-18:
@cslux I can absolutely assure you that the stock firmware could not cancel a print when printing from cura. I spent a lot of time with that code and it had no means to do so. Plus if my firmware can't do it then there is no other way to. It sends a host command to cancel job, so it just depends on whether or not cura listens for it.

Jyers, 2021-06-20:
Any update on this?

cslux, 2021-06-20:
@Jyers Well, now I'm unsure. I would need to flash the stock BL-Touch firmware and try it again. But I trust your words, you have modified the Marlin firmware and created an awesome Creality Ender 3 V2 firmware out of it, you know it better. Further, I didn't find anything which proofs I'm right in the Cura Repo (https://github.com/Ultimaker/Cura/blob/master/plugins/USBPrinting/USBPrinterOutputDevice.py), skimped a little through the issues, and source code but couldn't find anything.

Thank you!

Gil80, 2021-06-21:
I wasn't able to test it during this past week, so sorry for not having an update on my report.
I agree with @cslux this is an awesome firmware for our E2V2. Thanks for the firmware. I'll try to do more testing this week and provide an update.

Jyers, 2021-06-30:
Going to close this for now, If we get any updates feel free to post and I can reopen :)

closed 2021-06-30 by Jyers