Jyers/Marlin #1001: 1.3.4 -after filament change (pause at height from cura) nozzel goes inside the model and there is no play icon after filament change

darshan6rayate, 2021-06-13:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Jyers, 2021-06-13:
Try clearing eeprom by going to the control menu and selecting reset to defaults and then store settings.

Nothing has changed about how filament changes are handled between 1.3.2 and 1.3.4. So most likely this is and individual issue of some kind.

Jyers, 2021-06-20:
Closing this due to inactivity, if there are any updates or more information, feel free to post here and I can reopen this.

closed 2021-06-20 by Jyers