Jyers/Marlin #1002: Filament Runout Sensor still inverted after switching wires.

Sersch97, 2021-06-13:
### Description

The Filament Runout Sensor works inverted.
If there is no filament the printer prints without it. As soon as i put filament trough the sensor it triggers saying "filament runout".

I know it was already mentioned often here and you told us to switch the signal and ground wire. But no matter how i switch the wires it still happens or it doesnt do anything at all anymore. I also removed the + wire for the led and it still "works" saying "filament runout" when filament is in it without the led lighting. 

I removed the Red wire and checked the wiring. It was right. grnd from sensor to grnd from motherboard and signal from sensor to signal to motherboard.
Still happening. Then i switched grnd and singal and it was still happening. 

### Steps to Reproduce

1. [First Step]: connect filament runout sensor
2. [Second Step]: start a print
3. [third step]: wait to heat up and then after the first line it says "filament runout"

**Expected behavior:** after switching the wires it should be working not inverted anymore

**Actual behavior:** after switching the wire it still worked as inverted or not at all after trying to put the wires in any possible combination. After putting it back to the starting position it works again as inverted. 

#### Additional Information
Here are the Pictures of my sensor/switch.
[image] [image]

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
I use your 5x5 HS BLtouch 4.2.2 1.3.4 .bin

smooreace, 2021-06-13:
> [image: IMG20210613222759] <https://ibb.co/YWB7j90> [image:
> IMG20210613222806] <https://ibb.co/dcpn2dM>

Then compile the firmware with the properly set filament switch position

Jyers, 2021-06-13:
>Still happening. Then i switched grnd and singal and it was still happening.

You don't want to switch ground and signal, you want to switch it so that ground on the switch still goes to ground on the board, and Vcc (+) on the switch goes to signal on the board. This way the switch will pull signal to ground when triggered indicating a runout event. It is best to leave Vcc on the board disconnected.

smooreace, 2021-06-13:
Still infinitely easier to compile the firmware properly.

Sersch97, 2021-06-13:
Thanks jyers, it didnt work. It just did nothing then. I realised while testing that the printer is printing even if no sensor is connected. only if i have the sensor connected and it works as inversed it triggers the runout. but if i start the print with "filament sensor" activated but no sensor connected, the print prints as if there would be as sensor working normaly. the only thing i have connected is the wire from the motherboard but not sensor on it and it behave like there is one

Jyers, 2021-06-13:
That's because the firmware defaults to runout state low, which means nothing connected to the sensor port will register the signal pin an high, meaning the printer thinks everything is fine. Basically what you need to do is figure out what wire orientation will cause the signal pin to be pulled to ground when there is no filament in the sensor, when there is filament the signal pin should be floating since there is an internal pullup on the board.

Jyers, 2021-06-14:
Did you end up getting it working?

Sersch97, 2021-06-14:
> Did you end up getting it working?

hi,
sadly not. no matter how i connected the sensor it either didnt work at all or it just worked as inverse. 
thank you for the explanation of how it works in the firmware

Jyers, 2021-06-14:
That very strange, maybe try switching the wires on the filament sensor instead of on the mainboard?

Sersch97, 2021-06-14:
Well as i understood i had to connect 
ground from mainboard to ground from sensor
sensor from mainboard to + from sensor

is that correct?
because as i understood i shouldnt connect anything to + on the mainboard to prevent shortages
but that way its not doing anything
if i connect ground mainboard to + on the sensor and 
sensor from the mainboard to ground on the sensor it works as inverted.

gothcha, 2021-06-15:
No this will do nothing, the switch should have three connections, a common a NC and a NO. The ground on the board should go to the common of the switch and then either the NC or NO goes to the sensor on the board. This way the sensor is shorted to ground with either the switch open or closed. I hope this clarifies everything

Cal-LF, 2021-06-15:
I you look closely at the switch, the NC is connected to nothing on the white connector. Disconnect the NO switch terminal and jumper the NC to the NO circuit trace on the PC board.

Sersch97, 2021-06-17:
okay, i found some time to try out your suggestions.

what @gothcha said, helped me a lot. Thank you!
it now works properly. I have now connected board ground to common on the sensor and NC from the sensor to sensor the board. 

thank you again guys for the help and guiding me through.
i hope this issue might help others aswell

closed 2021-06-17 by Sersch97