Jyers/Marlin #999: 1.3.4 problem

Gitaryadeller, 2021-06-13:
### Description
selmalar benim birkaç problemim var ve yeni bir kullanıcı olarak sorun çözemedim

### Steps to Reproduce

1. [her yazdırma başlangıcında z offset 1 mm kadar yukarda başlıyor, 2. baskıya başladığım zaman ısıtma hatası yeniden başlat uyarısı veriyor ve sürekli tekrarlıyor 10 dakika bekledikten sonra hata vermeden ısıtma yapıyor özellikle ön ısıtmada sürekli yapıyor]

2. [tekrar başlattığım zaman düzeliyor]
3. [github da yeniyim bu pençere öğrenmem gerek bu kadar yazabildim kusura bakmayın]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Jyers, 2021-06-13:
Try clearing eeprom by going to the control menu and selecting reset to defaults and then store settings.

I haven't had any reports of similar issues, so if you get any more info or anything changes let me know. As it is I can't imagine what would be causing this.

Gitaryadeller, 2021-06-13:
I think the problem is in the blower fan, because ; When I give gcode start the blower fan does not work. the motherboard fan does not work and the drivers and mofset heat up early and give a heating fail error. z is surprised because the drive is getting hot. When I preheat, all the fans are active, but when I select gcode and start, the blower is not active. How can I activate it? When gcode starts in stock firmware, all fans work.

Gitaryadeller, 2021-06-13:
Note: The problem occurs when the device is hot. For example, I pressed the track, it is broken, I stop and restart. The problem starts right here, when the mofset starts to generate power again when it is hot, the mainboard fan stops, because the first layer blower fan is turned off, it causes early heating for a certain period of time, I think all fans should be active every time the printer starts.

Jyers, 2021-06-13:
Whether or not the fans are enabled right away when printing is determined by the gcode. If you want them to start right away, you can increase the initial layer fan speed in your slicer.

Gitaryadeller, 2021-06-13:
Ok I know. I want the fans to be turned off on the first layer. but at idle and preheat the fans should run, stop when i start typing and gradually speed up. Here my fans won't turn on until the bed and headboard are warm, shouldn't they run until I start typing and then stop when typing starts?

Jyers, 2021-06-13:
a heating failed error indicates that the hotend was not able to heat up fast enough and therefore triggered an error to be safe. Have you modified your hotend at all?

Gitaryadeller, 2021-06-14:
all stock exdruder bmg

Jyers ***@***.***>, 14 Haz 2021 Pzt, 00:52 tarihinde şunu
yazdı:

Jyers, 2021-06-14:
Try running a PID autotune. The menu for this can be found in Control>Temperature>PID

smooreace, 2021-06-14:
> I am using PETG filament, how many degrees do you suggest I set the pid
> settings

You shouldnt be "setting" them... you run the auto-tune as suggested

Jyers, 2021-06-14:
@smooreace I think they mean how hot to run the PID autotune at, you need to set the temp for it to cycle to.

@Gitaryadeller I usually run PETG at 220-230, but it depend on your filament. Just run it at whatever you normally print at.

smooreace, 2021-06-14:
I guess thats fair... I didnt read enough into it. Sorry

Gitaryadeller, 2021-06-14:
> @smooreace PID otomatik ayarını ne kadar sıcak çalıştıracaklarını kastettiklerini düşünüyorum, dönmesi için sıcaklığı ayarlamanız gerekiyor.
> 
> @Gitaryadeller PETG'yi genellikle 220-230'da çalıştırırım, ancak bu sizin filamentinize bağlıdır. Normalde yazdırdığınız şeyde çalıştırın.

The petg I use is 235 degrees, it is necessary to set the pid to 235 and I made it at 235. The jyers firmware also comes as 185. Does it cause a problem to increase the value in the stock firmware? There is no such problem in the stock firmware.

Jyers, 2021-06-14:
@Gitaryadeller Just to check, you did set it and then run it correct? Also the reason there is not an issue in stock firmware is because the thermal runaway settings are very lenient in the stock version. I modified them to improve the safety, but it does mean if your machine has trouble heating the hotend for some reason it is more likely to trigger.

Gitaryadeller, 2021-06-14:
> @Gitaryadeller Sadece kontrol etmek için ayarladınız ve sonra doğru mu çalıştırdınız? Ayrıca stok yazılımında sorun olmamasının sebebi stok sürümde termal kaçak ayarlarının çok hafif olmasıdır. Güvenliği artırmak için bunları değiştirdim, ancak bu, makinenizin herhangi bir nedenle hotend'i ısıtmada sorun yaşıyorsa, tetikleme olasılığının daha yüksek olduğu anlamına gelir.

yes, stock firmware is ok. I am currently using stock. I guess my need is for all the fans not to run before the printer starts writing gcode. Is there a way to do this? When I enable manual fan in Jyers firmware it is ok. The fan stops when I start typing. this is what it should be... by the way, your software is amazing, congratulations for doing such a work

Jyers, 2021-06-14:
Thank you :)

Let me just make sure I have this right. You want the fan to run when the printer is preheating before a print?

Gitaryadeller, 2021-06-14:
> Teşekkür ederim :)
> 
> Bu hakkım olduğundan emin olayım. Bir baskıdan önce yazıcı ön ısıtma yaparken fanın çalışmasını mı istiyorsunuz?

yes

Jyers, 2021-06-14:
I don't think that will help the issue here. If you are trying to combat the "Heating Failed" error, that means you need your hotend to heat up faster, enabling the fan won't help with that.

So did you actually run the PID autotune procedure? it should show a popup and take a few minutes

Gitaryadeller, 2021-06-14:
> Bunun burada soruna yardımcı olacağını sanmıyorum. "Isıtma Başarısız" hatasıyla mücadele etmeye çalışıyorsanız, bu, fanınızın daha hızlı ısınmasına ihtiyacınız olduğu anlamına gelir, bu da fanın bu konuda yardımcı olmayacağını gösterir.
> 
> Peki gerçekten PID otomatik ayar prosedürünü çalıştırdınız mı? bir açılır pencere göstermeli ve birkaç dakika sürmelidir

yes i do pid every time i load it. I just want to say that I am doing a pullback test for example, it takes 10 test prints to find this setting. a piece is finished, I start the test right after the device is hot, I print 1,2,3,4, the printer runs out, I start immediately. The motherboard stays hot when the stop-start is too much, and when I reboot, the blower fan and the motherboard fan stop while preheating. The mofset is already hot when the printer starts, the mofset is loaded for preheating. In the meantime, when the fans stop, the mofset may protect itself and stop heating. When I run the Jyers firmware printer and manually set the fan to 255 from the settings, it works during preheating and does not give an error. What I want to tell you is that when I start the printer instead of manually, its automatic operation will be more stable for motherboard cooling. you know the blower and the motherboard work the same and stop the same

Jyers, 2021-06-14:
Okay I think I get your meaning now. Sorry if I'm a bit slow to understand but the translation is not always the easiest to interpret. To achieve what you want you can put `M106 S255` before your preheat commands in your start gcode. This will run the fan at full speed during the preheat operation. You may also need `M106 S0` after to turn off the fans for your first layer. 

Alternatively, many people wire the mainboard fan to the hotend cooling fan terminals on the mainboard, so it always spins rather than being controlled by the parts cooling fan.

smooreace, 2021-06-14:
Im running mine straight from the power supply. There is never harm in too
much cooling, other than potential noise. And my fans are completely silent

Gitaryadeller, 2021-06-14:
> Tamam, sanırım şimdi anlamını anladım. Anlamakta biraz yavaş olduğum için özür dilerim ama çeviri her zaman yorumlaması en kolay olanı değildir. İstediğinizi elde etmek `M106 S255`için başlangıç ​​kodunuzda ön ısıtma komutlarınızın önüne koyabilirsiniz. Bu, ön ısıtma işlemi sırasında fanı tam hızda çalıştıracaktır. Ayrıca `M106 S0`ilk katmanınız için fanları kapatmanız gerekebilir .
> 
> Alternatif olarak, birçok kişi anakart fanını anakart üzerindeki hotend soğutma fanı terminallerine bağlar, böylece parça soğutma fanı tarafından kontrol edilmek yerine her zaman döner.

I may not have been able to explain it fully because I wrote it with a translation. yes, I want to separate the fan, I will do it parallel to the hotend fan. That's exactly what I meant, I'm new, I don't know much about coding but I understand what you're saying but I can do these in beginner gcode, thank you very much, I wish you continued success. it would be great to interfere with the retraction speed and distance setting on the printer, this is where I have the most difficulty, it would be great to intervene when writing a single piece to the printer, if possible jyers firmware goes to another dimension and opens an era... thanks for your help

Jyers, 2021-06-14:
Unfortunately without the use of experimental firmware retractions, there is no real way to control retractions after a print is sliced. I'm glad you're enjoying the firmware and that I was able to help with your issue. Happy printing! :)

closed 2021-06-14 by Jyers