Jyers/Marlin #1006: problema cordinate

acevux78, 2021-06-14:
### Description

### Steps to Reproduce

1. aggiornato alla versione 1.3.4 
2. le cordinate lampeggiano numero cordinata punto ? non capisco come mai prima non lo faceva quando metto home o qualsiasi altro punto come su manual livelling 
3. spero venga risolto il problema

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tititopher68-dev, 2021-06-14:
@acevux78 
Hello,
This is normal behavior. The coordinates flash when the printer no longer knows the exact position of the motor steppers. This is normal when the motor steppers are deactivated. These deactivate after a certain time. It will then be necessary to do a Home All again.

Jyers, 2021-06-14:
Just as @tititopher68-dev said, this is normal, and the reason you haven't seen it before is because it was introduced in 1.3.3. I'd recommend looking over the release notes when upgrading to avoid future confusion. 

> Added Coordinate Blink if not Homed

closed 2021-06-14 by Jyers