Jyers/Marlin #1019: [Feature Request] Save print time + filament consumption

Hologos, 2021-06-16:
### Description

Since the firmware now tracks how many prints have been printed and how much filament has been consumed, I'd like to save statistics to the SD card about every print. It would be great if includes print start (datetime), print end (datetime), total print time and filament consuption (length).

Thank you & be well.

tititopher68-dev, 2021-06-16:
@Hologos 
The system does not have an internal clock so the notion of date cannot be taken into account.
 On the other hand it is possible very easily to add a menu to save / load / reset the printing stats ...

smooreace, 2021-06-17:
Good info, but rendered useless the second you change filament. On the
other hand, Cura tells you how much time and filament each print will take.
Its pretty easy to track that across dozens of rolls of filament.

Hologos, 2021-06-17:
It doesn't make it useless for my use case which is getting paid for filament (+ el. consumption) for stuff my friends wants me print for them. Cure (and any other slicer honestly) is not that much acurate. Only the printer knows how much filament it pushed. The same goes to total print time.

Jyers, 2021-06-20:
@Hologos While you are correct slicers are not always accurate about print time, they are absolutely accurate as far as filament consumption. The slicer tells the printer when to extrude and how much filament to push, so except for some extensive circumstances with settings, the slicer will be very accurate about how much filament is being extruded.

As far as the feature in question, Marlin does not have any such implementation for such a thing so I can't do much there. I can however do as @tititopher68-dev mentioned and add a print counter reset if you really wish to see how much is extruded each print and are willing to reset it before every print.

closed 2021-06-20 by Jyers

Hologos, 2021-06-20:
I disagree that the slicer knows how much will be extruded in reality. If my e-steps are not set correctly, the comsuption will be off. But weighing the piece will do the same job.

Jyers, 2021-06-20:
Yes, I suppose incorrect settings is covered under "extensive circumstances with settings", but if your e-steps are off the printer will also not know how much is extruded, just how much it thinks it's extruding based on the set e-steps. So either way the slicer and printer will have the same value.

smooreace, 2021-06-20:
If you dont have things set up properly in the first place then there is no
point in pretending to collect the data. WAY too much effort for garbage
prints!