Jyers/Marlin #1025: BL Touch Not Adjusting Correctly--Inconsistent Z

HermanGaviriaMarin, 2021-06-18:
**Issue at hand** 
BL touch is successfully creating and accurate mesh of the build plate. However, when I go to print the X gantry is at inconsistent and erroneous heights as it moves around the build plate. Instead of creating a first layer that is consistent in height or "squish" (how close the nozzle actually resides to the bed) the nozzle is either extremely close or extremely far away from the build plate. A general occurrence from testing has shown (but is not limited to) , as an example, the left side is super close to the nozzle to the point of not allowing plastic to flow (or a microscopic layer) and the right side being so high up that the plastic is extruded into air. This isn't exclusive to right and left (sides of build plate), it seems that it is either not adjusting at all to the differences in height (recorded in mesh by BL touch)  or it is adjusting completely wrong. To clarify, I have placed G29 or M420 S1 in my starting G code (Cura 4.9.1)

**Background Info**
BL touch version 3.1 by AntLabs, main board version 4.2.2 from Creality. Stock Steppers all around with the exception of a secondary Z axis stepper running in parallel with original (both move at the exact same time with identical movement), to add on I have made sure that both sides (entire x gantry) or the x gantry are perfectly leveled with one another (how far up the left part of the gantry is compared to how far up the right side of the gantry is, this is done to make sure that both sides of gantry are at the same distance from build plate. All stepper connections have been checked and are working. 
- 4.2.2 board
- Tested on UBL 10x10, UBL 10x10HS, UBL 15x15HS, Jyers Config (self compiled), ABL BL touch 3x3, ABL BL touch 5x5. 
- Same Problem across mentioned Firmware 
- All metal hot end
- Noctua Fan Upgrades (PSU, Mainboard, Hotend) 
- Octoprint enabled (used in tandem with printing from SD card) 
- Printing Bed is Creality stock glass
- Yellow springs (this is what I first tested with first) then I used silicone spacers
- Probe Offsets x34 y -57

**Steps to recreate**
1. Flash any of the previously stated Firmwares
2. Have and BL touch version 3.1 by Antlabs installed (installed into correct dedicated port on 4.2.2 board)
3. Level X gantry by turning lead screws (if using double z steppers), match left and right sides with caliper by taking heights from top of X gantry v extrusion (20x20) to the frame extrusion on the bottom (40x40).
4. Tram your build plate (level it), But since we're using BL touch there shouldn't be a need for it but I still did it. And have tested both leveled and unleveled.
5. Pre heat bed and nozzle to regular printing temperatures (for its 210 for nozzle and 60 for bed)
6. Create a new mesh using LCD display. Select: Level>Create New Mesh> Save Mesh to Eprom. Lastly Enable the option called "Leveling Active "
7. Prepare a Bed Leveling test STL (will be attached)
8. Turn STL into Gcode then print.
9. During Print adjust Z offset to get desired "squish"
10. Inspect During print and post print whether or not the first layer is all at the same height. 

**What I've tried to fix the issue**
- Leveled bed
- Leveled X gantry heights
- Changed firmware 
- unplugged and plugged BL touch back into place (both on mainboard and on BL touch)
- Changed Starting Gcode from "M420 S1" to G29 and vice versa
- Printed from octoprint and from SD card

**BL Touch amazon link:**
https://www.amazon.com/Creality-BLTouch-Upgraded-Leveling-Mainboard/dp/B08L9DHP5R/ref=sr_1_6?dchild=1&keywords=bl+touch+ender+3+v2&qid=1623997130&sr=8-6#customerReviews

**Thoughts and Conclusions**
My BL Touch has worked in the past without a problem, this problem arose recently. Since the BL touch creates and records mesh without a single problem I can't imagine how its a problem with hardware (but that's why I'm here, because I'm not 100% sure). Hopefully its just firmware, would hate find out that the BL touch is what's causing the issue. 
[Bed Leveling Test.zip](https://github.com/Jyers/Marlin/files/6675094/Bed.Leveling.Test.zip)
[image]
[image]
[image]
[image]
[image]

tititopher68-dev, 2021-06-18:
@HermanGaviriaMarin 
Sorry, but looking at your Mesh, ouhlala, there is a serious problem.  Having a standard deviation of 1.6mm largely explains your printing problems.  Compensation can never be done correctly under these conditions.  For example, you go over 1 or 2 cm to -0.5 or + 0.5mm !!
 Such a distorted glass plate, I can't believe it.  I'm more thinking of a BLTouch pb or a Bed problem (part in Alu, by the way do the following test: remove the glass plate, take a precision ruler, place it on the edge in different place of the Bled and check  if you can fit between the bed and the ruler, 1 or 2 or 3 sheet of paper).  If so, contact Creality and they'll change the Heated Bed for you without any worries.  If this is your glass plate, I advise you to buy a mirror plate (to be cut in 235x235mm).  It's perfectly flat, costs next to nothing, and it works great.  If it is the BLTouch, do the M48 precision test, in the menu and check the deviation which should not be greater than 0.002mm ...
 By way of comparison, with my original bed, In UBL, I have a maximum deviation of 0.1mm .... with my mirror, of 0.008 mm.
 Good luck...

Cal-LF, 2021-06-18:
You have a hardware problem.

smooreace, 2021-06-18:
Several really

HermanGaviriaMarin, 2021-06-18:
The mesh you saw was the one created using the silicone bed spacers. Though I got the same results with the yellow springs even after leveling perfectly 

[image]

Jyers, 2021-06-20:
@HermanGaviriaMarin When you say "leveled perfectly" how does your mesh look? I'm not trying to question you leveling, but as has been mentioned, the mesh you posted in your picture is quite severely out of level. Almost 1.6mm off from left to right. This is going to be almost impossible to properly compensate for with ABL. After a good level your mesh should generally be within 0.1-0.2 mm edge to edge.

If you have done this and are still having issues we can dive deeper into the issue so we can help you get this fixed, but I can almost guarantee this is an individual issue of some kind rather than an issue with the firmware as a whole.

HermanGaviriaMarin, 2021-06-20:
Grandmaster Jyers, I usually think I do just fine, as of yesterday I kinda gave up and leveled manually. That manual level was the closest thing to perfect, I will create a mesh of that current tram and see whether or not it is or isn't working. Tbh since no one else is having this issue I'm really thinking it must be an issue on my hand. Regardless I will keep you updated. I'm fairly new with 3D printing and still got ways to go before I can determine problems as accurate as you veterans.

mikeyboy49, 2021-06-27:
I have similar issues with the BLTouch and levelling, my mesh around a glass plate is between 0.1 to -O.4 but I couldn't get a a test pattern to be correct all over. I have since stripped my printer cleaned everything, made sure it's all square. Testing this week.

krisr, 2021-06-27:
I'm also struggling to get a consistent "squish" across all corners of my bed with mesh leveling enabled. In my case, the bed has been leveled well manually with < 0.1 mm variance between the two extremes.

Is there a way I can more clearly observe the z compensation at different points on the grid during a print to verify it is applying correctly? I've tried to observe it manually but I don't see any movement of the z axis while printing the initial layers.

I'm confident that I have my starter gcode configured correctly (G29 comes right after G28), but no matter what I do I can't get consistent squish when printing mesh leveling squares. It's as if mesh leveling isn't compensating at all. I followed the guide here and did every step correctly to my knowledge:
https://smith3d.com/ender-3-v2-bltouch-firmware-installation-guide-by-smith3d-com/

I'm using this bl touch kit from creality:
https://www.amazon.com/dp/B08PTWDVVS?psc=1&ref=ppx_yo2_dt_b_product_details

I've tuned my z offset such that the middle, top left, and bottom right squares prints perfectly but the top right and bottom left appear under extruded no matter what I do.

I get consistent z offsets for each of the 9 corners. I am heating the bed before running g29 and before the initial leveling.

Thanks for any help.

krisr, 2021-06-27:
I believe I already answered my own question. To verify that leveling was working correctly, I added a business card to one corner, re-ran leveling so it thought the mesh was higher there, then I set z to 0.2 and used the controls to move the head to each corner of the grid with leveling enabled. I could clearly see that mesh leveling compensation was on because the nozzle raised above the business card in the corner where it had seen it while leveling, and the business card did not fit in other corners.

I suspect my problem is a function of bilinear sampling not having enough points to adequately compensate for the differences in my bed leveling rather than bed leveling itself not being activated during printing.

grilparto, 2021-06-30:
Maybe you folks can try something out here to see if you're having the same problem as #1036?  Try the following steps to see if you get a noticeably different mesh.

- Before doing anything Reset to Defaults.
- Create a new mesh.
- View the mesh/take a picture of it for reference.
- Change the steps/mm of one of the axis under Configure -> Motion -> Steps/mm.  (Maybe try setting X from 80 to 82).
- Create a new mesh again.
- View the mesh/take a picture of it.  Compare to the reference from before.

Jyers, 2021-07-09:
Going to close this for now. If any new information comes up that could point to this being an issue with this fork specifically we can reopen this. Otherwise I would recommend opening an issue on the main marlin repo if the issue persists.

closed 2021-07-09 by Jyers