Jyers/Marlin #1030: [Feature Request] Preview

NyxCode, 2021-06-19:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

gothcha, 2021-06-19:
This has been raised before see #885 #862

Jyers, 2021-06-19:
I'll leave this here so I don't forget to look into this. I'll have to see if any other marlin screen implementations have accomplished this and how they went about it...

tispokes, 2021-06-27:
As far as i know there is a cura plugin for creating these previews/thumbnails. Working on ender3. But not on v2, I've v2, so I'm not able to verify this.

Tw0bit, 2021-07-01:
The fork of Jyers for the Voxelab - ALEX [ https://github.com/alexqzd/Marlin/releases/ ] - appears to have something going for Prusa as in this video -https://www.reddit.com/r/VoxelabAquila/comments/o9zyqz/just_a_little_sneak_peek_of_something_im_working/

Jyers, 2021-07-03:
@Tw0bit Yeah, I'm actually working with Alex to get this going

Tw0bit, 2021-07-03:
Nice work. I have both an Anquila and Ender 3V2 if you need any testing, etc.

Webcr0ler, 2021-07-27:
Hi, perhaps you can drop an eye here,
He seems to have already done that.
https://www.reddit.com/r/Ender3v2Firmware/comments/ombgj1/do_you_want_more_advanced_features_on_your/
https://github.com/mriscoc/Marlin_Ender3v2/releases/tag/2.0.9.X

Jyers, 2021-07-27:
I've seen that implementation, and while it's functional, what me and Alex are cooking up should be a bit cleaner and give some more options.

dante-35, 2021-12-08:
I've finished merging this repo and alex's previev feature after 2 days of work :).
i made a pull request. It works great for now but ony tested couple of hours. I dont know if it has a bug or a major issue.
You can check  my repo but  i am not an experienced programmer.
Thats why i dont recommend it before @Jyers or another experinced programmer finishes checking.