Jyers/Marlin #1035: Thermal Runaway after updating to 1.3.4

N3tYaroz3, 2021-06-20:
This is what happens.
I start a print from octoprint, the bed and hotend heat up, but the temperature is not what I set in Ultimaker Cura.
The temperature rises to 260 °, I manually set the temperature to 240 ° (PEtG mold) and after a few seconds it starts to sound and the printing stops. This has never happened to me, I don't understand if it's a problem with the new update.

This is octoprint's message

Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Thermal Runaway, system stopped! Heater_ID: 0 - echo:THERMAL RUNAWAY

smooreace, 2021-06-20:
To me, temps should always be defined in the slider. Everything about 1
specific slice is centered around a target filament. So there is zero logic
in not also attaching the temperature settings.... but

Did you run a PID auto-tune?

N3tYaroz3, 2021-06-20:
> Per me, le temperature dovrebbero sempre essere definite nello slider. Tutto ciò che riguarda 1 fetta specifica è centrato attorno a un filamento target. Quindi non c'è logica nel non allegare anche le impostazioni della temperatura .... ma hai eseguito un auto-tune PID?
> […](#)

Hi, thanks for answering me. I did the autotune PIDs today. I don't know how to attach the temperature settings

Jyers, 2021-06-20:
I'm not sure I fully understand why your slicer temp doesn't match the temp you want to print at, but yes if you change the temp midway through heating it may create a thermal runaway error as the printer has trouble understanding what has happened and why the temps are so far off. There's not much I can do to correct this, it's simply the price to pay for good safety measures.

closed 2021-06-20 by Jyers