Jyers/Marlin #1055: [Feature Request] Shutdown RPi/OctoPrint from the printer's LCD menu

JMVS, 2021-06-24:
### Description

Have the ability to shutdown RPi/OctoPrint from the LCD

### Steps to Reproduce

There's an OctoPrint plugin already available that could take the M118 echo and execute a command (like shutting down).
Action Command: https://github.com/benlye/OctoPrint-ActionCommandsPlugin

**Expected behavior:** [What you expect to happen]

Select an option in the printer's LCD menu to have the RPi/OctoPrint to shutdown grecefully.

**Actual behavior:** [What actually happens]

None. Not implemented.

#### Additional Information

This would be advantageous for those that have the RPi connected directly to the printer (like myself). It could also be expanded to other commands available with the abovementioned plugin.

Jyers, 2021-06-24:
Hmmm, I definitely like the idea of this. I'll look into it!

Jyers, 2021-08-15:
Check out the latest commit 😉

closed 2021-08-15 by Jyers