Jyers/Marlin #1057: [Manual Leveling] not using full bed size

jolau, 2021-06-24:
### Description

I'm using V1.3.4 and when doing manual leveling, it doesn't use anymore the full bed size. I.e. the center point is not anymore at the center. And top left is not fully in the top left corner. I suspect it is because in the Configuration.h the bed size is now only (since commit 9a9a2655dfcac94209f2bcfe9b656dacb0b5ae82):
```
// The size of the printable area
#define X_BED_SIZE 200
#define Y_BED_SIZE 200
```

### Steps to Reproduce

1. Go to Manual Leveling Menu
2. Select Center Point

**Expected behavior:** Move to center point

**Actual behavior:** Moves not to center point but to point deviated in left and bottom direction

Cal-LF, 2021-06-24:
That bed size is from the default Marlin config. Jyres uses 230 x 230 for all of the templates. I you use the 3x3 manual mesh, the bed size is 230 x 230.

Jyers, 2021-06-24:
Are you talking about manual mesh leveling or manual tramming (4 corner) In either case, you don't want the points to be all the way in the corner, it is more productive to have them closer to the bed screws

closed 2021-06-24 by Jyers

jolau, 2021-06-25:
I'm talking about the manual leveling. If you compare bottom left to bottom right position, it is clearly visible that the bed size is wrong. And also the center position is not in the center.
[image]
[image]
[image]
[image]

mikeyboy49, 2021-06-25:
Looks like you have an offset, did you choose the right firmware for manual bed levelling (No BLTouch) .. just a suggestion.. I have similar issue but with BLTouch fitted..

Cal-LF, 2021-06-25:
Is the bed size set to 230 x 230 in the slicer?

Cal-LF, 2021-06-25:
I just used my manual level and all five points are correct. Using the 3x3 manual mesh template.
Try reinstall firmware and clear eprom.

Jyers, 2021-06-25:
Try clearing eeprom by going to the control menu and selecting reset to defaults and then store settings.

Ive seen issues like this before caused by old eeprom.

Also, if you're compiling your own firmware you should not use the default config in the marlin folder. Use the Default-NoProbe set found in Configuration Files/E3V2 Templates

mikeyboy49, 2021-06-25:
I used the 3x3 BLTouch config file and made adjustments to the BLTouch for Z pin and a couple of other items. It works but have I done wrong in using that file?? Will it cause me further issues once I start printing , I only did it yesterday and not really use it much?

smooreace, 2021-06-25:
You probably should use either UBL-NoProbe or Manual Mesh. Kinda silly to
use a config that starts off being configured completely wrong for your rig

mikeyboy49, 2021-06-25:
I do have the BLTouch but the config code wasn’t set for the Z pin, it was just a case of changing 2 lines in the code. I was just concerned if there was anything else I should worry about, I realised that I actually posted on here but it didn’t really concern me , sorry

smooreace, 2021-06-25:
Oh... I got names all screwed up. Sorry. Thought your were the original poster.

You can also verify that your probe offset is correct and apply your Z-Offest. But those things can also be adjusted via menu and GCODE after the fact