Jyers/Marlin #1060: M600 - Continue from OctoPrint does not dismiss dialog on printer

NguyenTanBao, 2021-06-24:
### Description

I'm testing out M600 from GCode. After changing the filament, there're pops up on both Printer and OctoPrint to ask if I need to purge more. I used OctoPrint and select Continue, print continues but the purge more dialog is still showing on the printer

### Steps to Reproduce

1. Create a test gcode file with M600.
2. Print, change filament
3. Select "Continue" on Purge More dialog on OctoPrint

**Expected behavior:** [What you expect to happen]
Purge More dialog on printer should also auto dismissed.

**Actual behavior:** [What actually happens]
Purge More dialog still shows on printer.

#### Additional Information

Cal-LF, 2021-06-26:
Reinstall firmware and clear eprom.

Jyers, 2021-06-27:
Hmmm, the firmware might actually not know how to deal with that. I'll take a look and see if I can sort it out.

Jyers, 2021-06-27:
Alright, this should now be fixed with the latest commit

closed 2021-06-27 by Jyers