Jyers/Marlin #1061: M600 - reheat time out while changing filament

NguyenTanBao, 2021-06-24:
### Description

Reheat timeout while changing filament

### Steps to Reproduce

1. Create gcode with M600
2. Print and changing filament when prompted

**Expected behavior:** [What you expect to happen]
There should be no reheat timeout problem.

**Actual behavior:** [What actually happens]
Reheat timeout while changing filament.

#### Additional Information
* OctoPrint 1.6.1
* E3V2-UBL-NoProbe-5x5-v4.2.2-v1.3.4.bin
* DWIN_SET 1.3.2

Cal-LF, 2021-06-24:
Suggest clearing eprom. Timeout issue was resolved with the V1.3.4 release.

NguyenTanBao, 2021-06-25:
I'll give it a try but this is too inconvenient, there are other data saved (like UBL mesh, manual levelling).

smooreace, 2021-06-25:
Its not inconvenient at all... its essentially required when you update
firmware. You cant expect to totally replace an existing foundation and
think that everything depending on that foundation wouldnt be compromised!

Jyers, 2021-06-27:
@NguyenTanBao Any luck with the reset?

Also UBL mesh is not affected by reset, any any other settings can simply be restored using an M503

closed 2021-06-30 by Jyers