Jyers/Marlin #1065: Heating failed errors occur when changing target temps after a preheat

MohnJadden, 2021-06-25:
### Description

Heating failures occur when adjusting temps after preheating bed

### Steps to Reproduce

1. Preheat PLA or ABS
2. Let the bed temperature increase to its preheat value
3. Change the hot end temp to any value higher than the current hot end target temp, or change the bed temp to any value lower than the current bed target temp
4. Wait approximately 30 seconds

**Expected behavior:** The printer should adjust the temps to the specified targets

**Actual behavior:** Printer gives a heating failure error, beeps loudly, and requires a physical power off/on, regardless of temp

#### Additional Information

* I'm using the stock 1.3.4 release from E3V2-ManualMesh-5x5-v4.2.2-v1.3.4.bin 
* The issue occurred on 1.3.3 as well - it doesn't happen on Creality official firmware

smooreace, 2021-06-25:
I go through that process almost daily. Ive never seen this issue.

mosheas, 2021-06-25:
Yes, I do. Exactly the same problem.

tititopher68-dev, 2021-06-25:
@mosheas @MohnJadden 
I never seen this issue too...
Hardware issue...?

mosheas, 2021-06-25:
> @mosheas @MohnJadden
> I never seen this issue too...
> Hardware issue...?

I didn't see it with stok firmware

tititopher68-dev, 2021-06-25:
@mosheas 
yes, but with the stock firmware, unlike this firmware version based on the near latest Marlin bugfix versions, you don't see thermistor errors. I advise you to do an PID Autotune of the Bed and the Hotend on at least 8 cycles. It can also improve the situation ;-)

Taffse, 2021-06-26:
I had the same problem, and did what  @mosheas suggested but no luck. So I reinstalled the firmware and installed the stock firmware for the screen and did the PID one more time. Now everything is ok. I will try to upgrade the screen software ones more but i'm printing now so. Thanks for the tips and thank you Jyers for a great update ;)

smooreace, 2021-06-26:
Screen firmware is... for the screen. Obviously, that has zero connection
to the heating process

Taffse, 2021-06-26:
Just telling what i did to get it to work smooreace.... So, the solution for me was to reinstall the firmware. Thanks for pointing that out :)

Jyers, 2021-06-27:
@MohnJadden 
Let me know if @Taffse's solution works

In the meantime I'll try to recreate this as I feel like I've heard of this issue before.

agarthand, 2021-06-28:
Fresh user of this great Marlin fork. Thanks for all your efforts!

I'm able to reproduce this issue on my machine as well. Did not occur everytime but most of them. I'm still trying to figure out if it only happens in case I change the temperature due the display, as I can not remember that this happen when I change temperature through Octoprint. 

Installed latest releases of Jyers Marlin FW and the gotcha icon DWIN_SET.

smooreace, 2021-06-28:
And again... did you do a PID auto-tune after clearing the EEPROM after
installing the new FW?

agarthand, 2021-06-28:
Not yet, but it's on my list to be done next and be assured I'll provide an update if it fix the situation for me or it reamins unchainged.

agarthand, 2021-06-30:
Update, I performed a PID for hotend and bed but the kill exception still occurs from time to time.
In the actual case I aborted a print (sent through Octoprint) - because it starts sliding - and immediately after the abort was performed I initiated a pre-heat of bed and hotend (via Octoprint). After a few second it starts yelling and requires me to shut it off. Seems to happen if the actual temperature is still close to the target (actual 210°C / target 235° hotend; not sure about the actual bed value, target for bed was 75°C).

phodara, 2021-06-30:
I have exactly the same problem.  My steps to recreate:
1. Heat the bed
2. When the bed reaches temperature, turn off the heaters
3. Level the bed with Bed Visualizer
4. Print the job
*5. As the bed heats for the print job  printer crashes with heat failure and load beeping.
6. I power off printer and power back on.
7. Now I can print the job without a problem

I am using OctoPrint 1.6.1, Python 3.7.3, OctoPi 0.18.0 and Marlin Jyer v1.34
This happens every time.

Gitaryadeller, 2021-07-01:
> @MohnJadden @ Taffse'nin çözümü işe yararsa
> bana bildirin
> 
> Bu arada, bu sorunu daha önce duymuş gibi hissettiğim için bunu yeniden oluşturmaya çalışacağım.

yes, we talked to you about this problem before, the same problem happened to this friend and a friend on facebook had the same problem and returned to stock software, it gives a heating error while heating and says start again when it starts to heat, it gives the same error only when the device is cold, there is no error

phodara, 2021-07-02:
I don't understand, what is the cause and the solution?

Tw0bit, 2021-07-02:
I have this when the print temp is lower than the current temp. 

IE if the hotend/bed are at say PETG temps of 235/70 and then the sliced file is for PLA 195/60, when the GCode changes to the lower temp it crashes with the error.

Provided the pre-heated temp is at or below the desired print temp no error.

phodara, 2021-07-02:
Hmmm interesting I will try to verify tomorrow.

phodara, 2021-07-02:
That seems to work. Provided I only prewarm to a lower temperature than what I print at, no crashes. Probably a minor bug in the firmware.  As long as there is a work around it's fine with me.

Jyers, 2021-07-03:
I don't know if it's actually a bug. My best guess is that resetting the temp to make it cooler is messing with marlins thermal runaway safety system. As cooling the heaters would cause the temp to change to slowly and therefore trip the runaway protection. I could make the runaway values even more lenient but obviously I don't want to overdo it and compromise the safety of the machine.

phodara, 2021-07-03:
Once I preheat I turn the heaters off before I run the print job. I don’t
know the thermal runaway logic, but once a print job starts and the gcode
presets the temps but the current temps are higher shouldnt PID kick in and
keep the heaters off and monitor temp rates until preset temps are reached?
My understanding was thermal runaway triggers only when temps go above
preconfigured maximum determined temps.  Again I don’t know anything about
the logic I am just speculating.

I really do like this firmware and the screen layout, very intuitive and
very stable and great features.

agarthand, 2021-07-04:
Is someone able to check / reproduce the following behaviour, I just recognized?

Running into Thermal runaway kill when I initiate a print with a 10° higher temperature for the first layer(s), once it changes down to the regular temperature.

Configured values 245°C for initial 2 layers, then continue with 235°C. As far as I see the kill occurs once it changes set temperature to 245°C but is not yet cooled down to this value (showing actual of about 239°C).

I mean yes actual temp is higher then the (new) set value, but does this mean it is not possible to have a 10° higher temp for initial layers? Or need the runaway protection to be a bit more patient is such situations?

phodara, 2021-07-04:
I just had another thermal runaway and the hotend temp was 5 degs cooler than what the print job called for. Once again a reboot fixed the problem.  I will test agarthand issue ,an increased temp for initial layer, within the next day or two.

samirthegreat, 2021-07-07:
I am also facing the same issue. The error occurs if the target temperature is close to current temperature. Suppose I preheated PLA then clicked on cooldown and after 1 min if I preheat PLA again I get the error. Sometimes it also occurs before printing.

phodara, 2021-07-07:
Only solution I have found is to reboot printer then let print job bring up temperature

samirthegreat, 2021-07-07:
> Only solution I have found is to reboot printer then let print job bring up temperature

But that is not exactly a solution right? You have to wait for your printer to cool down at least 10°C on the bed and heat up again. 
Mriscoc firmware doesn't have that problem. 
https://github.com/mriscoc/Marlin_Ender3v2/wiki/Color-Themes?fbclid=IwAR0iE7ivn7KSTw2fCmDXoCfY6h8-YbYlOtrJUNz1nTRN-xtdgdiptAzF8KE

phodara, 2021-07-07:
Yes not a fix, It is a workaround. I am not familiar with Mriscoc. It appears to give you a lot of control of the colors in the GUI.  What I really like about the Marlin Jyers build is all the advanced options such as PID autotune and many others see: https://github.com/Jyers/Marlin/wiki/Update-Notes
In the past if have been using the the smith2d flavor of marlin but it looks like they dropped theirs in favor of jyer's because its better and more up to date. Time will tell.

Jyers, 2021-07-07:
The reason Mriscoc doesn't have this problem is his fork still uses the stock thermal runaway settings, which are very loose and not the safest. However I believe all of this can be fixed by tuning them slightly without putting them all the way back to stock. I'll give it a try and anyone here compiling their own version can weigh in.

Jyers, 2021-07-09:
I've tuned the settings a bit now. Let me know if anyone compiling their own can still manage to recreate the issue with these new settings.

Cal-LF, 2021-07-09:
@Jyers These are the settings I use and have no thermal error.
225 #define THERMAL_PROTECTION_PERIOD 30        // Seconds
245 #define WATCH_TEMP_PERIOD 30                // Seconds
246 #define WATCH_TEMP_INCREASE 3               // Degrees Celsius
253 #define THERMAL_PROTECTION_BED_PERIOD       180 // Seconds
259 #define WATCH_BED_TEMP_PERIOD               180 // Seconds

Tuncay-Ayhan, 2021-07-10:
I also use those numbers for the temps, except for the bed. No more thermal
runaway ever since.

samirthegreat, 2021-07-11:
> I've tuned the settings a bit now. Let me know if anyone compiling their own can still manage to recreate the issue with these new settings.

Thank you so much! I think the issue is solved.

phodara, 2021-07-11:
I am not sure this is the appropriate place to ask for help, if not please let me know.
This is my first experience with platformio I am trying to compile the jyers temperature-issue-testing codebase so I can make the temperature adjustments and I am having a a couple of issues.
I believe I have everything installed correctly including all support software Atom, VSC, PlatformIO with marlin-addon etc. I did watch the recommended video.
I have downloaded the jyers temperature-issue-testing codebase and moved the header files to their appropriate location.  I set the default_envs = STM32F103RET6_creality.
but when I compile I get a lot of errors  with the last notable error:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:77:4: error: #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."

To confirm I have everything installed correctly I downloaded the marlin nightly bugfix code base and when I go to build I have two environment options:
STM321F103RET6_creality
STM321F103RET6_creality_maple

If I use STM321F103RET6_creality this fails but if I use STM321F103RET6_creality_maple the build is successful.
I am not sure what the ramifications of these two different environments are and why I don't see them with the temperature-issue-testing codebase, and why I get the errors with the temperature-testing codebase.
Any assistance would be appreciated.
Thank you.

samirthegreat, 2021-07-11:
> I believe I have everything installed correctly including all support software Atom, VSC, PlatformIO with marlin-addon etc. I did watch the recommended video.

Uninstall everything install VSCode then install platformIO extension then install Auto Build Marlin Extension. Open the folder in VSCode. Copy the appropriate config.h and config_adv.h files. then go to Auto Build Marlin tab and press build. It will compile a bin file.

phodara, 2021-07-11:
K will try now.  Thanks

phodara, 2021-07-11:
I am copying the  header/config files from  /configuration files/jyers's Config  to /Marlin and getting the same  errors:

In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:72:4: error: #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
   #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:77:4: error: #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
   #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_flash.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:72:4: error: #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
   #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_flash.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:77:4: error: #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
   #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:72:4: error: #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
   #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:77:4: error: #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
   #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_bl24cxx.cpp:29:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:72:4: error: #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
   #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\eeprom_bl24cxx.cpp:29:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:77:4: error: #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
   #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:72:4: error: #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
   #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:32:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:77:4: error: #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
   #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\MarlinSerial.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:72:4: error: #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
   #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
    ^~~~~
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\MarlinSerial.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:77:4: error: #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
[... cut ...]
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\Servo.cpp.o] Error 1
*** [.pio\build\STM32F103RET6_creality\src\src\HAL\STM32F1\SPI.cpp.o] Error 1
============================================================================================== [FAILED] Took 13.58 seconds ============================================================================================== 

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:00:13.578

Cal-LF, 2021-07-11:
Atom extension is not used.
Also follow the instructions to install "development mode" in the text file. This will eliminate other errors.
[VS Code Development.txt](https://github.com/Jyers/Marlin/files/6797667/VS.Code.Development.txt)

Cal-LF, 2021-07-11:
You can not use Jyres config files. They are for his particular setup. The E3V2 templates are the correct config files to use.

phodara, 2021-07-11:
Hmm the one I downloaded was Marlin-Temperature-Issue-Testing.zip.  I just downloaded Marlin-Ender_3_V2_JyersUI.zip is this the correct one?

Cal-LF, 2021-07-11:
Yes, the default on the top of the list.

Cal-LF, 2021-07-11:
It may be better to open a discussion topic rather than be off topic in this thread.

phodara, 2021-07-11:
Thank you @Cal-LF  it worked!  :-)

Cal-LF, 2021-07-11:
Just as a reference, these are my changes from default for my configs. They are not all required for my printer but set some defaults and different temps for safety.
[Jyres Compile Changes.txt](https://github.com/Jyers/Marlin/files/6797705/Jyres.Compile.Changes.txt)

phodara, 2021-07-11:
Thnx

Jiriteach, 2021-07-27:
Is there a new build with this fixed? Or does one have to manually compile?

I was experiencing this daily which because really annoying as I moved temps between preheat and ideal quite often. 

Am now using mriscoc and don’t see this issue. That’s running very well but would like to come back to using Jyers. 

Thanks

Cal-LF, 2021-07-27:
@Jiriteach There is no new release but it should be soon. Mr. Myers has done a lot of merge cleanup and adjusted the temp settings for heating failure.

phodara, 2021-07-27:
I did a manual compile and It worked flawlessly.  I no longer have temperature issues.
This was my first compile and there is plenty of docs how to do this online.

Jyers, 2021-07-27:
@Jiriteach Yes, there should be a release soon. I've had my hands a bit full with #22422 but I'll try to get a new release out today or tomorrow.

@phodara Glad to here it's working well! Seems like we've finally got it tuned for max safety with little to no false trips.

Jiriteach, 2021-07-27:
Awesome! Thanks all and to @Jyers  for the excellent work on this. Looking forward to the new release.

agarthand, 2021-08-02:
Just installed the new release and for it seems to fix the problem. After several tries I was not able to reproduce my issues any more. Thanks for the adjustments. Will monitor behavior closely for the next prints and let you know if anything similar occurs.

Jyers, 2021-08-02:
Great! I'll mark this as closed for now. Hopefully we've got it sorted 🥳

closed 2021-08-02 by Jyers