Jyers/Marlin #1067: [Feature Request] allow firmware flashing via OctoPrint

datapaganism, 2021-06-25:
### Description

It would be nice if the firmware was enabled so we could try to flash the firmware using OctoPrint,
I have lost all of my sdcard adapters.

Sorry if this has been raised before

### Steps to Reproduce

to quote from [here](https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/marlinbft.md)

### Enable the Binary File Transfer protocol

To compile firmware with the binary file protocol enabled, uncomment this line in `Configuration_adv.h`:

`#define BINARY_FILE_TRANSFER`

### Set SDCARD_CONNECTION to ONBOARD
The SD card connection must be configured for `ONBOARD`.  If it is set to `LCD` the firmware file will be copied to the SD card in the LCD and the board will not update when it is reset.

To set the SD card to `ONBOARD`, modify the `#define SDCARD_CONNECTION` line in `Configuration_adv.h`:

`#define SDCARD_CONNECTION ONBOARD`

tititopher68-dev, 2021-06-25:
@datapaganism 
Interesting but not really possible with the Ender3V2.  Why ?  Because the SDcard must be formatted before process flashing for Creality V4.2.2 and V4.2.7 motherboards ! Which means that you have to use an adapter...What the plugin does not allow...

datapaganism, 2021-06-25:
I only made the feature request after seeing that people [could do it themselves](https://www.reddit.com/r/ender3v2/comments/nq89lf/how_to_easily_flash_a_firmware_with_octoprint/), The way I use is my printer is that is hooked up to OctoPrint 24/7 and the SD card is only inserted after a new firmware release.

adam3654, 2021-06-25:
It can be done, I haven't flashed firmware manually in a while now. [See this discussion](https://github.com/Jyers/Marlin/discussions/539#discussioncomment-634957)

tititopher68-dev, 2021-06-25:
@datapaganism @adam3654 
You get confused: writing data to the SD card (which is indeed possible) and flashing the firmware after writing the file to the SD.
 Firmware must be written to a Formatted SD card before!  However, you cannot format the SD via Octoprint!

datapaganism, 2021-06-25:
@tititopher68-dev why would you have an unformatted SD card in your printer in the first place?
Now I'm confused

tititopher68-dev, 2021-06-25:
@adam3654 @datapaganism 
I complete by saying that this method can be used on other motherboards (BTT or MKS or even the Aquila Voxelab) which do not require SD formatting ... but not with Creality v4 motherboards.  2.2 or v4.2 7 which require the prior formatting of the SD ... therefore to format with an adapter the card on the PC ...
 On the other hand, I do not know if it is possible to flash the firmware directly from the PC as with the CR10 V2 for example.

tititopher68-dev, 2021-06-25:
> @tititopher68-dev why would you have an unformatted SD card in your printer in the first place?
> Now I'm confused

Because to flash the firmware on Ender3V2, the MicroSD card must be formatted before...

datapaganism, 2021-06-25:
@adam3654 I take it you have compiled your own version of the firmware then, I guess I shall do the same

adam3654, 2021-06-25:
> @adam3654 I take it you have compiled your own version of the firmware then, I guess I shall do the same

Yes, there are good step by step guides that makes setting up the environment quite easy. Then you just have to make edits to 2 text files as desired, and click a button in the "auto build marlin" extension. Then upload the bin file to the printer using the octoprint extension (there's a setting in octoprint that needs to be modified that is covered in the link I provided earlier to avoid problems) and you are all set.

smooreace, 2021-06-25:
Firmware most definitely does NOT have to be put on a formatted card. I have NEVER used a clean card, and flashed dozens of times

tititopher68-dev, 2021-06-25:
@smooreace:
Maybe but I have already had problems doing as you do and in the WIKI, as elsewhere, it is strongly recommended to reformat the SD to avoid any possible problem ...

@datapaganism @adam3654 : OK, you don't necessarily need an SD to flash the motherboard. I modified my build derived from that of Jyers to have a clean process.
I took the opportunity to add a very useful option concerning end of filament detectors ....

smooreace, 2021-06-25:
I think that the reason for the recommendation is to prevent the ignorant
from trying to flash with multiple .bin files on the card. That step alone
probably prevents several dozen support "calls" a month, in turn saving a
fortune in man-hours. There is zero logic to it NEEDING to be done
otherwise. Thats simply not how data works.

closed 2021-06-26 by datapaganism