Jyers/Marlin #1070: Gotcha icons

wolfeman2323, 2021-06-26:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

smooreace, 2021-06-26:
Absolutely!

tititopher68-dev, 2021-06-26:
@wolfeman2323 
A joke !!!????

wolfeman2323, 2021-06-26:
I don't understand this reply.  Thought this was a forum for help.  You
people are just PTWs.

Cal-LF, 2021-06-26:
If you are having an issue, it would help to state what it is. Or maybe it belongs in the discussion section.
Either way, learn to communicate.

smooreace, 2021-06-26:
The ultimate troll! post nothing, then bitch when people are blind to the importance of nothing. Classy!

tititopher68-dev, 2021-06-27:
@wolfeman2323

GordinhoPl, 2021-06-27:
> ### Description
> ### Steps to Reproduce
> 1. [First Step]
> 2. [Second Step]
> 3. [and so on...]
>
> **Expected behavior:** [What you expect to happen]
>
> **Actual behavior:** [What actually happens]
>
> #### [...]

We don't even know what is wrong, what is the issue. You just paste the template to report bugs, but you don't fill the description, so sry, but nobody will help if we don't know the issue...

tititopher68-dev, 2021-06-27:
@GordinhoPl 
He's just a person who wants to make his sound interesting ... You shouldn't answer him again until he makes a serious request!

closed 2021-06-30 by Jyers