Jyers/Marlin #1075: Scrolling sometimes fail

tispokes, 2021-06-27:
### Description

Scrolling sometimes do not react propertly. E.g. scrolling thru printable files or in the menu.

### Steps to Reproduce

1. have some more files on the sd card
2. to go "Print" menu
3. try to scroll down a bit faster

**Expected behavior:** Scroll correctly

**Actual behavior:** sometimes nothing, you have to scroll click by click.

Jyers, 2021-06-27:
Looks like it's struggling due to the amount of files on there. Did it work better with stock firmware? If so I'll look into how I might be able to optimize the handling for such large quantities of files.

tispokes, 2021-06-27:
Due to the other sorting on stock fw i don't have to scroll that far. newest file last is may bit contraintuitive on this kind of using.
If you want, i will go back to stock and test it out. okay, i'll do.
But whats about "newest file first"?
Thx for the fast reply and this incredible good fw upgrade. Would like to donate you.

tispokes, 2021-06-27:
Tested it quickly. The stock fw (v1.1.1) just show a amount of older files and then stops. You are not able to scroll any further.

Jyers, 2021-06-27:
Oh so it limits how many files you can scroll through? That's very strange...

Also this firmware sorts alphabetically as that was the marlin default. I just tried recent first and it made scrolling painfully slow. It might take some kind of overhaul to get that working decently

tispokes, 2021-06-27:
Hm when you watch my video, your fw seems not to be sorting alphabetical.
If i have enough spare time, I'll will investigate this problem a bit.
Is there a possibility to save the settings onto the sd card and reload them after a firmware update?

I'm new to your firmware, just installed it some hours ago.

smooreace, 2021-06-27:
You can via plugin in OctoPrint

tispokes, 2021-06-27:
Nice thanks. At the moment I'm printing w/o octo cuz i had trouble in the printing quality. Using raspi 3 from 2016. But perfect for this usecase maybe.

tispokes, 2021-06-27:
> You can via plugin in OctoPrint
> […](#)

Nice thanks. At the moment I'm printing w/o octo cuz i had trouble in the printing quality. Using raspi 3 from 2016. But perfect for this usecase maybe.

Jyers, 2021-06-27:
M503 also outputs the commands you need to restore settings

Jyers, 2021-07-09:
From what I've seen on other issues it seems the slowness of scrolling stems from the current handling of sd card files. Hopefully this will get changed soon with some new work me and Alex are doing.

closed 2021-07-09 by Jyers