Jyers/Marlin #1082: Probe Failure with E3V2-BLTouch-5x5-v4.2.2-v1.3.4

Fribb, 2021-06-28:
### Description

When I use the E3V2-BLTouch-5x5-v4.2.2-v1.3.4.bin on my Ender 3v2 I can't run a full cycle of the ABL because one of the 25 Points will result in a Probe failure, cancelling the print.

I tried a different version (E3V2-UBL-BLTouch-10x10-v4.2.2-v1.3.4) and the Bed level cycle went through 200 probes without failing even once.

I reverted back to the previous version (because the UBL resulted in the corners being printed correctly but the centre had gaps on each line) and started the same model again and after the 6th probing it failed again. After restarting the print job it went through without a problem.

### Steps to Reproduce

1. Flash E3V2-BLTouch-5x5-v4.2.2-v1.3.4.bin
2. add the G29 T Gcode
2. start a print 
3. wait for probe to fail

**Expected behavior:** Probing fails rarely but not with every new print at least once

**Actual behavior:** Probe fails with each new print at least once

gothcha, 2021-06-28:
See #192 for info on BLTouch probe failures and fix using either X_min port or adding pull-up resistor.

Fribb, 2021-06-28:
> See #192 for info on BLTouch probe failures and fix using either X_min port or adding pull-up resistor.

Interesting read but I think you mean Z_MIN here?! At least that is what is referenced in the solution discussion.

Still, why is there such a difference in behaviour in the UBL and ABL firmware that I tried out? 200 probes without issue with UBL and the first cycle of ABL in a 5x5 grid fails.

gothcha, 2021-06-28:
Oops yes Z_Min, it is not a firmware issue, so almost certainly just a random failure mode. The only reliable solution is as discussed unfortunately.

smooreace, 2021-06-28:
I have inconsistent failures too. I can do 20 3x3 scans sometimes and never
have an issue. But doing 10x10s I have to power-cycle the printer between
scans to have a chance at success. Doing 3 back to back sometimes gets as
far as the 3rd point and it stops

Jyers, 2021-06-30:
Yes, unfortunately this is most likely a hardware issue. If you get some more info and feel you may have evidence contrary to that, you can update this or better yet bring it to the marin marlin repo where that info would be more valuable

closed 2021-06-30 by Jyers