Jyers/Marlin #1094: Issue related to the width of Fonts

tititopher68-dev, 2021-06-30:
When displaying numeric values of type abc.dd (with 2 decimal places) and especially with the new fonts
The number "a" is altered. The effect is even more flagrant when it comes to a negative value -abc.dd or the number 9, 8, 6 (for "a")

To fix it:
Either reduce the width of the digits effectively -> I posted a PR to @Quint097
Either modify the firmware code as follows :

void CrealityDWINClass::Draw_Float(float value, uint8_t row, bool selected/=false/, uint8_t minunit/=10/) {
uint16_t bColor = (selected) ? Select_Color : Color_Bg_Black;
if (isnan(value)) {
DWIN_Draw_String(false, true, DWIN_FONT_MENU, Color_White, bColor, 194 , MBASE(row), F(" NaN"));
} else if (value < 0) {
DWIN_Draw_FloatValue(true, true, 0, DWIN_FONT_MENU, Color_White, bColor, (minunit == 100) ? 6-log10(minunit) : 5-log10(minunit), log10(minunit), 202, MBASE(row), -value * minunit);
DWIN_Draw_String(false, true, DWIN_FONT_MENU, Color_White, bColor, (minunit == 100) ? 186 : 194, MBASE(row), F("-"));
}
else {
DWIN_Draw_FloatValue(true, true, 0, DWIN_FONT_MENU, Color_White, bColor, (minunit == 100) ? 6-log10(minunit) : 5-log10(minunit), log10(minunit), 202, MBASE(row), value * minunit);
DWIN_Draw_String(false, true, DWIN_FONT_MENU, Color_White, bColor, (minunit == 100) ? 186 : 194, MBASE(row), F(" "));
}
}

Jyers, 2021-06-30:
@tititopher68-dev I'm sure that change works well, but I should really make the formatting and negative sign dynamically space with the number of digits anyway. I'll get on that.

tititopher68-dev, 2021-07-01:
@Jyers 
I did the formatting and negative sign dynamically space with the number of digits anyway...The result is satisfactory and works with the different FONT PACKs.

code : 

void CrealityDWINClass::Draw_Float(float value, uint8_t row, bool selected/*=false*/, uint8_t minunit/*=10*/) {
  uint16_t bColor = (selected) ? Select_Color : Color_Bg_Black;
  uint16_t ndigits = 0;
  if (!isnan(value)) ndigits = ((value == 0) || (abs(value) < 10)) ? 1 : floor(log10(abs(value))) + 1 ;
  if (isnan(value)) {
    DWIN_Draw_String(false, true, DWIN_FONT_MENU, Color_White, bColor, 194 , MBASE(row), F(" NaN"));
  } else if (value < 0) {
    DWIN_Draw_FloatValue(true, true, 0, DWIN_FONT_MENU, Color_White, bColor, (minunit == 100) ? 6-log10(minunit) : 5-log10(minunit), log10(minunit), 202, MBASE(row), -value * minunit);
    DWIN_Draw_String(false, false, DWIN_FONT_MENU, Color_White, bColor, 220 - (6 * ndigits), MBASE(row), F("-"));
  }
  else {
    DWIN_Draw_FloatValue(true, true, 0, DWIN_FONT_MENU, Color_White, bColor, (minunit == 100) ? 6-log10(minunit) : 5-log10(minunit), log10(minunit), 202, MBASE(row), value * minunit);
    DWIN_Draw_String(false, false, DWIN_FONT_MENU, Color_White, bColor, 220 - (6 * ndigits), MBASE(row), F(" "));
  }
}

Jyers, 2021-07-01:
I'm mostly done with fixing it but thanks. If my method doesn't work I'll use yours, they're basically the same just a few different ways of doing things.

Jyers, 2021-07-03:
I just pushed the commit for dynamic float sizing. If you could check to make sure it works with the new fonts that would be great. I managed to simplify the code a good bit and avoid all the ternary madness ;-)

tititopher68-dev, 2021-07-03:
@jyers
There are math errors in the code.  A value of 0 is not possible, because log10 (0) is infinite ....
Likewise if the value is 0.05 this returns for digits the value 0 instead of 1 ....
Finally, if in a menu, you have values ​​to display with 1 and 2 decimal places, the display of the values ​​is shifted, they do not line up on the point and it is not very aesthetic.

tititopher68-dev, 2021-07-03:
@Jyers 
I just tested in "real" now...I am quite amazed at the behavior of the system with a "log10 (0)"....
Finally, your code works ... Just a little thing to modify to avoid a "blinking or flicking" effect : 
modify just : 
DWIN_Draw_Rectangle(1, Color_Bg_Black, 194, MBASE(row), 236 - (digits * 8), MBASE(row)+16);
by : 
DWIN_Draw_Rectangle(1, Color_Bg_Black, 192, MBASE(row), 232 - (digits * 8), MBASE(row)+16);

and it should be perfect !

Jyers, 2021-07-03:
Great, I'm just a bit confused on that edit... I know the blinking your talking about, but I had already fixed it by making it 134, not 136. Are you saying with with new fonts that it still blinks with 134?

Jyers, 2021-07-03:
And yes, I could have error corrected log10, but it seems to do so fine on its own. Also the reason you may have thought that the code would have failed for minunit 100 is a slight misunderstanding of how the digits value is created. It adds one if there is a decimal (ie minunit > 1) and also adds digits for the amount of decimals (ie minunit 100 adds 2 extra digits) That way it is able to properly space everything no matter the number.

tititopher68-dev, 2021-07-03:
@Jyers 
I tested all the FONT and with:
DWIN_Draw_Rectangle (1, Color_Bg_Black, 192, MBASE (row), 232 - (digits * 8), MBASE (row) +16);

This also corrects with some FONT the appearance of a blue line with a width of 2 points when going from a digit of 4 digits to 3 digits or from 3 digits to 2 digits ... and no more Blinking , the display is very clean !

Regarding  to "log10 (0)", in C ++, it depends on the system on which the code is executed. It seems that the Ender3V2 or even the Aquila doesn't care; -) ....

closed 2021-07-05 by Jyers