Jyers/Marlin #1095: [feature request] filament detection state and hotend temp

Knotty91, 2021-06-30:
I’m wondering if it’s possible to have an option in the advance menu to change the filament detection state. What ever style filament sensor is installed, you would be able to change the detection, either high or low, without having to modify the firmware. 

Also another good advance menu item would be to have the option to change the max temp of the hot end. So if you were to install a different hot end you could get more temp out of it. I would think to have it maxed at the capabilities of the thermistor. 

Thanks

Cal-LF, 2021-06-30:
Been discussed further down just recently.

Jyers, 2021-06-30:
Unfortunately both of these are made impossible (at least without messing with how marlin functions) by the fact that marlin defines these settings on compile, making them immutable after the fact.

One alternative for the runout sensor issue is to switch the wires as is discussed well here: #981

closed 2021-06-30 by Jyers

Knotty91, 2021-07-01:
Thank you for the explanation