Jyers/Marlin #1096: Creality E3V2 Touchscreen

Legsmaniac, 2021-07-01:
Is there any way this firmware will work or can be made to work with the new Creality Touchscreen?

https://www.creality3dofficial.com/collections/accessories/products/upgraded-touch-panel-screen-display-for-ender-3-pro-ender-3-v2

I have tried it on my Ender 3 V2 with the Creality Touchscreen firmware and it works OK but the firmware is awful.
If I can't use Jyers or similar firmware, it's going back for a refund and I'll stick with the knobby original. 

I tried uncommenting #define DWIN_CREALITY_TOUCHLCD in Configuration.h on the off chance it worked but sadly not.

It's a nice wee screen but it's not that important.  Certainly not if I can only use Creality's firmware over gool ol' Jyers.
Advice/help appreciated.
Thanks.

Jyers, 2021-07-01:
I have a facebook post somewhere where I explained this in depth, if I find it I'll link it here. Long story short the touchscreen displays use a very different system of interfacing with the mainboard and require a lot more programming of the screen, so while I may attempt to support it at some point, it most likely will not be anytime in the near future.

closed 2021-07-01 by Jyers

Legsmaniac, 2021-07-01:
Jason, thank you.  That's exactly what I needed to know and pretty much what I feared.
No worries though, I'll send it back for now and get a full refund, no problems.
I'll stick with the knobby box, it's works fine after all.  Even more so with the best firmware loaded.  😉
Keep up the good work and thanks again.

Dids, 2022-01-03:
@Jyers Did you ever find that Facebook post?

I'd be interested in getting this working, as well as putting in as much work & effort as required, however I'm definitely not familiar with the differences between the two displays (yet).

I'm sure others would gladly pitch in too, one way or another, as it's a relatively popular part/upgrade at the moment.

DerKnerd, 2022-02-25:
I might necro bump it, but how is the situation with the touchscreen? I would hate to go back to the rotary knob display to use your firmware :slightly_smiling_face: