Jyers/Marlin #1098: print head does not lift when canceling printing

madorax251, 2021-07-01:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Jyers, 2021-07-01:
Do you have your "on print cancel" gcode script set up to lift the head?

adam3654, 2021-07-01:
> Do you have your "on print cancel" gcode script set up to lift the head?

That would only help with cancelling from Octoprint. Having the nozzle pull away from the print while the nozzle is still hot would be nice.

Jyers, 2021-07-01:
Canceling from the display does cancel from octoprint. When you select cancel the printer sends a command to octoprint telling it to cancel, just the same as if you had canceled directly from octoprint.

adam3654, 2021-07-01:
I did not know that (obviously ;-) ), thanks for the info. I just chased through the source code to learn about the host actions. Very cool!

madorax251, 2021-07-02:
> Canceling from the display does cancel from octoprint. When you select cancel the printer sends a command to octoprint telling it to cancel, just the same as if you had canceled directly from octoprint.

madorax251, 2021-07-02:
@Jyers 
I am using GCODE Scripts which is listed in the octoprint settings on your wiki page

madorax251, 2021-07-02:
what gcode script should i use in octoprint settings?and gcode for bed visualizer

closed 2021-07-02 by madorax251

madorax251, 2021-07-02:
@adam3654 
> Itu hanya akan membantu membatalkan dari Octoprint. Menjauhkan nosel dari cetakan saat nosel masih panas akan lebih baik.

I've tried canceling printing via Octoprint printhead not picking up

Jyers, 2021-07-03:
G91
G0 Z10

That should lift the print head for you. For more info on gcode check out https://marlinfw.org/meta/gcode/