Jyers/Marlin #1099: Jyers 1.3.4 - Can Home but no other move (level, etc) commands exectute

Tw0bit, 2021-07-02:
### Description
Running an Ender V2 with the Jyers 1.3.4 ( E3V2-BLTouch-3x3-HS-v4.2.2-v1.3.4.bin ), previous was 1.3.2 (E3V2-UBL-BLTouch-10x10-HS-v4.2.2.bin). After flashing the firmware, did a reset defaults.

Intermittently the printer head does not move when commanded by the screen for manual level, z-offset and mesh manual leveling. However auto home works. It works for a while then oddly stops working until a reboot.

### Steps to Reproduce
1. Upgrade from 1.3.2 to 1.3.4
2. Use printer
3. After a time, the head no longer moves.

**Expected behavior:** [What you expect to happen]
The head to move

**Actual behavior:** [What actually happens]
The head doesnt move

#### Additional Information
Video of the issue for manual bedlevel / z-offset. Shows homes but not move. The display that normally says "wait" flashes then stops -- https://drive.google.com/file/d/1KNGYIpk88VkQ3OeKquDeCIFfBGC7VqCX/view?usp=sharing

How can I do a full EEPROM wipe? It seems in another issue that the mesh for UBL stays after resetting defaults in the non-UBL firmware.

smooreace, 2021-07-02:
Cant see much of anything in the video, as you can see.

Tw0bit, 2021-07-02:
I did test the G-Drive link as public, here is the download link (112mb) - https://drive.google.com/uc?id=1KNGYIpk88VkQ3OeKquDeCIFfBGC7VqCX&export=download

chrisneal, 2021-07-02:
@Tw0bit You can do a full ERPOM clear with:

```
M502 ; Load defaults
M500 ; Save
M501 ; Load saved defaults from EPROM
```

Tw0bit, 2021-07-02:
@chrisneal thanks, I assume that is what the Jyers UI runs in the background; but I have used the M commands. Noting after doing this, I still had the weird issue as posted above.

Jyers, 2021-07-03:
Very strange indeed. Perhaps a hardware issue of some kind, as I can't imagine what else could cause such behavior. Overheating comes to mind...

Tw0bit, 2021-07-03:
Its very strange, never had it on the 1.3.2 build that i ran continuously for about 6weeks. I did an 8hour print last night no issues.

This only pops up when I am messing around with the bed leveling. I assume possibly the BLTouch maybe no firing at some point and locking the request/skipping it. Anyway to debug easily?

Jyers, 2021-07-03:
Unfortunately not that I can think of. If you think it's the BL Touch you could try running without it.

Jyers, 2021-07-05:
Any updates on this?

closed 2021-07-09 by Jyers