Jyers/Marlin #1100: Unclear mesh viewer values

Qubi-B, 2021-07-02:
I've fallen, and turns out not only me, that the black color on the mesh viewer represents warped bed. [here](https://www.reddit.com/r/ender3v2/comments/obiuxe/nooo_glass_bed_cant_be_warped_until_you_use_bec/h3ql2u6/) is Reddit comment under my post with the photo that shows people's confusion. My suggestion is- to make values display + and - and to make colors easier to understand. (This is only a suggestion, that may improve user experience IMO)

Jyers, 2021-07-03:
The main reason the viewer doesn't have + and - is because it doesn't fit with larger meshes. But you're right it's a bit unclear. I'll add the colors to the legend to people understand green is positive, red is negative.

Qubi-B, 2021-07-05:
Thx, closing

closed 2021-07-05 by Qubi-B