Jyers/Marlin #1107: Please rebase :)

dwery, 2021-07-03:
Would be nice to have a rebase, thanks!

etwasmitbaum, 2021-07-03:
I am sorry but I am not familiar with the term "rebase" here.
Do you mean to update to the latest Marlin version?

dwery, 2021-07-04:
> I am sorry but I am not familiar with the term "rebase" here.
> Do you mean to update to the latest Marlin version?

https://lmgtfy.app/?q=rebase

etwasmitbaum, 2021-07-04:
Excuse my stupidity, but thank you anyway.

dwery, 2021-07-04:
> Excuse my stupidity, but thank you anyway.

You aren’t stupid and that wasn’t a stupid question. Just a lazy one. It works much better if you ask questions along the lines of “I researched rebase on Google and on the Z forum and I found this link and that link but I have still some doubts regarding I and J”.

etwasmitbaum, 2021-07-04:
>You aren’t stupid and that wasn’t a stupid question. Just a lazy one. It works much better if you ask questions along the lines of “I researched rebase on Google and on the Z forum and I found this link and that link but I have still some doubts regarding I and J”. It shows that you have done your homework and that you care. People will be much more inclined to help.

Thanks for the info :).
I'll keep that in mind.

Jyers, 2021-07-05:
I assuming you mean a merge/pull from the bugfix branch. Trust me, you don't want me to do a rebase as it would mess up everyone's local repos...

But yeah, I'll pull from upstream soon.

Jyers, 2021-07-05:
Done :)

closed 2021-07-05 by Jyers