Jyers/Marlin #1108: Filament runout message freezing

ZDITTRIC, 2021-07-04:
runout sensor message is freezing. I have a BTT smart sensor that is showing filament runout (filament is present but that’s a separate issue - runout distance set to 12.5mm), once message comes up I hit cancel to resume print, it says filament runout and stays indefinitely. Pre configured UBL 15x15

### Steps to Reproduce

1. [First Step] Runout sensor trips 
2. [Second Step] hit cancel after purge to resume print 
3. [and so on...]

**Expected behavior:** [What you expect to happen] should resume print

**Actual behavior:** [What actually happens] freezes on runout message

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead use one of the following options:

- The Marlin Firmware forum at https://reprap.org/forum/list.php?415
- The MarlinFirmware Facebook Group at https://www.facebook.com/groups/1049718498464482/
- The MarlinFirmware Discord Server at https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the latest "bugfix" branch to see whether the issue is already addressed.

-->

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Cal-LF, 2021-07-04:
Update to latest FW version.

HermanGaviriaMarin, 2021-07-04:
Yea I run into this issue a lot.  Only way to leave this unending magekyo sharingan isanami is to use octoprint and click continue. Without it there's no way to continue the print

tititopher68-dev, 2021-07-04:
@ZDITTRIC 
First  : modify your configuration.h file for BTT SmartSenor : //#define FILAMENT_MOTION_SENSOR by : #define FILAMENT_MOTION_SENSOR and check if you have : #define FIL_RUNOUT_STATE     LOW
Second : Compile your own firmware with the latest version.

ZDITTRIC, 2021-07-04:
I’ve updated to 1.3.4, the runout state is low and the issue persists. False trigger is an intermittent issue for me, with it happening after a few hours of printing.

smooreace, 2021-07-04:
That sounds more like a problem with your sensor @ZDittric. The firmware
really doesnt have a level of "interpretation" that it applies. It simply
reacts as you tell it to, to an input you told it to look for.

ZDITTRIC, 2021-07-04:
> That sounds more like a problem with your sensor @ZDITTRIC. The firmware really doesnt have a level of "interpretation" that it applies. It simply reacts as you tell it to, to an input you told it to look for.
> […](#)

I agree, it is likely a hardware issue. It is bringing emphasis to the screen freeze issue though

Cal-LF, 2021-07-04:
Is the sensor close to the extruder? Too far away will cause false trip. Screen freeze is likely a hardware problem. Grounding and shielding may be needed.

ZDITTRIC, 2021-07-04:
Sensor is about 25mm from extruder, mounted to the gantry wheels

Cal-LF, 2021-07-04:
My runout length is set at 10mm. I get false trips at 7mm as recommended in the instructions for the BTT.

ZDITTRIC, 2021-07-04:
I have mine set at 22mm. Don’t get as many false trips, only when printing at high speed (100mm/s setting). 

What do you think needs to be grounded to avoid the screen freeze?

Cal-LF, 2021-07-04:
I had random menu screen flips while printing. After doing the Creality grounding recommendation, they stopped.
Also ran a separate ground wire to a nearby conduit.
[image]

ledcomp, 2021-07-04:
By running a separate ground wire to a conduit could create a ground loop that could cause more problems than it solves. The grounded cord plugged into a grounded outlet is all that is needed to dissipate any built up static or noise.

Jyers, 2021-07-05:
Just for a but off clarification here, where are you hitting cancel? When runout occurs it should show a runout popup, and then a filament change popup after it's successfully parked, and then a purge more popup after your filament change. After purge more it should have a resuming print popup. At what point in the process are you being sent back to the runout popup?

ZDITTRIC, 2021-07-05:
It shows the purge more pop up, where I hit cancel to resume the print. After this it pops up filament runout and stays indefinitely.

Cal-LF, 2021-07-05:
@ZDITTRIC That action did happen in V1.3.3 and was corrected in V1.3.4. Are you sure you have the latest Version installed?

Jyers, 2021-07-09:
@ZDITTRIC After the purge more popup it should go to a resume print popup, if it put you straight back to filament runout that either sounds like an issue with how marlin is handling it or an issue with your sensor itself.

Jyers, 2021-07-21:
Any update on this?

tungquach165, 2021-07-25:
I had the same problem in 1.3.3 and now it's the same in 1.3.4. Notice of "Filament runout" as soon as printing starts!

ZDITTRIC, 2021-07-25:
> Any update on this?

No issues after switching to UBL 15x15 non high speed. Switched to high speed and had issue again.

tititopher68-dev, 2021-07-25:
What is certain is that it is not a firmware problem, but a hardware problem only and certainly different depending on the cases described here.
 Note that some v4.2.2 motherboards have a problem with the BLTouch 5v power supply (5 pin port), which can cause the BTT Samart sensor to malfunction, since the 5V power supply is common.
 There are also sometimes misalignments of the photoelectric cell of the sensor, which is corrected elsewhere or clogging of the cell.
 Finally maybe also problems of mass and main power ...

tungquach165, 2021-07-25:
> What is certain is that it is not a firmware problem, but a hardware problem only and certainly different depending on the cases described here.
> Note that some v4.2.2 motherboards have a problem [...]

Hello, if it's hardware why it doesn't happen with old firmware but only with this one. I have used a HS 5x5 version of Smith3d before and didn't have any problems?

tititopher68-dev, 2021-07-25:
Because :
1- The 2 firmwares are not based on the same version of Marlin!
 2- The Runout is managed by the Marlin base code only, in the firmware just a graphic popup to inform the user and nothing else.
 3- I have one and I have no problem !!!

smooreace, 2021-07-25:
...and every precompiled bin file is built from potentially thousands of
completely different variables. The only way you will know is by doing some
proper troubleshooting to figure out exactly what is wrong. But "firmware"
in general isn't it.

AlphaDE, 2021-07-31:
I had the issue just a few minutes ago with code base 1.3.5, luckily I am using Octoprint, so I could resume from there.

Jyers, 2021-07-31:
I'll have to see if I can recreate this

tititopher68-dev, 2021-08-01:
@Jyers 
I have for my E3V2 a BTT SmartSensor and for the Aquila a classic runout sensor and never had this issue.

AlphaDE, 2021-08-01:
And to amend, I am using a BTT, too on a Ender 3 V2 4.2.7 mainboard.

badbod, 2021-08-10:
it is very common with these printers for the ground not to make it beyond the PSU, due to paint etc. getting in the way.  All owners of this printer should ensure to run ground wires from frame to PSU and to all steppers, as also recommended by creality. If you have a ground spike then ALSO ground spike the frame.

 Proper earthing is a must and you not to rely solely on the earth in the mains plug for it.  This may or may not solve some issues with electrical noise but is absolutely required for your safety.   If you experience any issue regarding 'ground loops' then have the earth in your electrical supply checked and repaired immediately as any such problem shows your electrical earth is not properly earthed.  You can have as many earths as you require for your own confidence and additional safety.

ZedJaved, 2021-09-15:
I have this same issue using a UBL-BLTouch-10x10 firmware with smart filament sensor enabled. Sometimes it works fine and other times it just freezes showing the filament runout message straight after selecting cancel on the purge option. I believe it's happened after selecting confirm on the purge popup too.