Jyers/Marlin #1115: build error -- : 'class CardReader' has no member named 'startFileprint'

dwery, 2021-07-05:
Marlin/src/lcd/dwin/creality_dwin.cpp: In member function 'void CrealityDWINClass::Print_Screen_Control()':
Marlin/src/lcd/dwin/creality_dwin.cpp:4936:20: error: 'class CardReader' has no member named 'startFileprint'
 4936 |               card.startFileprint();
      |                    ^~~~~~~~~~~~~~
Compiling .pio/build/STM32F103RET6_creality/src/src/lcd/dwin/rotary_encoder.cpp.o
*** [.pio/build/STM32F103RET6_creality/src/src/lcd/dwin/creality_dwin.cpp.o] Error 1

darco415, 2021-07-05:
same error after a new clone repo

tititopher68-dev, 2021-07-06:
@dwery 
The latest Marlin updates don't add much functionally to our Ender3V2s, but do bring a lot of variable naming or function naming in the code.  The compilation woes are certainly not over to readjust the ton of parameters with these changes of nomination.
 Your outcome is a perfect example.  The function just changed its name: startOrResumeFilePrinting () but it does exactly the same as startFileprint () ...
 And for 1 month, every week, Marlin has had fun doing this ... complicating the maintenance of all the forks.  This explains that many delay updates as long as they do not bring real corrections or new features for the printer in question.

Jyers, 2021-07-06:
Yup looks like I've got some merge conflicts to clean up. I'll get it fixed up

Jyers, 2021-07-07:
Reverted for now as I'm gonna need some time to fix all the issues

closed 2021-07-07 by Jyers

tititopher68-dev, 2021-07-09:
As for Marlin, somewhere they criticize Jyers' choices and applaud those of mriscoc.
In fact Jyers' firmware is quite easy to read and understandable and instead of valuing this work, Marlin prefers complexity and no longer wishes to see any other path than theirs ... (I noticed, coincidence or coincidence) that since a few weeks Marlin has been changing the base function name or variable name in the code without actually modifying the actual function, which makes it more and more complicated to update the Marlin base code for Jyers, an indirect and subtle way of making people understand to fall into line ?...

Excerpt from the conversation : MarlinFirmware#21942

**"_There seem to be a lot of coders messing around with this display because it's accessible, and this is resulting in fragmentation

I have been warning various parties about this from day #1. Over many months, very little has been upstreamed into Marlin, so I applaud @mriscoc for this work on getting the code into Marlin - as eventually, the forks get too far apart and too difficult to keep in sync.

This can be seen already with the Jyers fork

It would be much better to combine efforts than the existing paths of everybody going their own way - as eventually, everyone's fork gets too far to bring back - no matter the intentions._"** Comment approved by : Scott Lahteine - Project maintainer for MARLIN !

@Jyers,
I support all the work done and to come ... However, I come back to the choice to leave EXTUI ... If my hunches are correct, it might be better to create some kind of JyersLCD in Extui and be less dependent on Marlin modifications, much like BigtreeTech does. Obviously, Marlin does not intend to follow your path at all and favors that of mriscoc. It would be a shame ... I also think that it would be interesting to use the GETTEXT functions for the text displays, even if it means adding the missing messages in the language_en.h and other databases. This would allow you to reduce the size of the code once compiled but above all to open up the interface to other languages ...

Jyers, 2021-07-09:
@tititopher68-dev Scott chimed in a little more open-mindedly than CRC after

I did a quick rebase of the Jyers fork on top of the current bugfix-2.0.x code to see how many changes are required. Of course, it is quite different from this PR and the current bugfix code, but it probably contains all the functions that this PR has. The simplest way to compare is to flash each one and look at the user interface in action, trying out all the functions, and if anything is missing or incomplete compared to the other, then bring over the implementation.

dwery, 2021-07-10:
One thing is fundamentally true: the more outdated a fork becomes, the harder is to bring it up to date and to eventually merge. I remember someone in the discussione was working on a callback based approach to add menu items. That would make it much more maintenable.