Jyers/Marlin #1116: BL touch flashes when z axis homes

peterADM, 2021-07-06:
### Description
I recently got a Bl touch 3.1 for my ender 3 v2 from creality and a creality 4.2.7 main board, i installed it with correct wiring and unplugged the old z endstop. I saw that this firmware has lots of features for the bl touch and i wanted to try it. I flashed the mainboard with the "E3V2-BLTouch-3x3-v4.2.7-v1.3.4", but when i have to home the z axis the bl touch flashes and i thought maybe the lcd needs to be flashed so i did but i could only find the dwin lcd for a 4.2.2 i flashed the lcd and it worked, i tried and tried again, flashing and flashing but it does not want to home the z axis.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]
to home the z axis
**Actual behavior:** [What actually happens]
the z axis moves to the correct height , then the probe starts flashing red in 1-1.5 sec intervals until the home is canceled
#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Jyers, 2021-07-06:
Just to check, is your BL Touch connected to the dedicated BL Touch port, or the endstop?

peterADM, 2021-07-06:
yes its on the dedicated bl touch port

Akujiproxy, 2021-07-07:
I'm running into this same issue, the bltouch port often doesn't work for the Z-axis. Apparently theres a resistor issue on some boards? Anyways, I solved it by using the z-endstop port instead. Was using Smith3d firmware but wanted to switch to this latest Jyers build only to find out that there is no precompiled firmware using a 5x5 HS mesh with Z-endstop switch. Now I don't mind compiling my own but I'm not sure what settings are used in the other precompiled variants.

smooreace, 2021-07-07:
There is a set of pre-defined config files in the download. It's rather
easy to compare any of those files to a default config file to quickly see
the differences between them. That will lead you straight to what
you're looking for

peterADM, 2021-07-07:
> I'm running into this same issue, the bltouch port often doesn't work for the Z-axis. Apparently theres a resistor issue on some boards? Anyways, I solved it by using the z-endstop port instead. Was [...]

I cant do that the cable that come with my bl touch is the 5 pin connector.

Akujiproxy, 2021-07-07:
> I cant do that the cable that come with my bl touch is the 5 pin connector.

You sure the wiring is correct? Do the colors match the ones in this screenshot?

https://i2.wp.com/smith3d.com/wp-content/uploads/2020/08/7.jpg?w=1000&ssl=1

I read on multiple forums that orange and yellow are reversed on some cables and it messes things up.

peterADM, 2021-07-07:
> You sure the wiring is correct? Do the colors match the ones in this screenshot?
>
> https://i2.wp.com/smith3d.com/wp-content/uploads/2020/08/7.jpg?w=1000&ssl=1
>
> I read on multiple forums that orange [...]

ill try to use smith 3d but i like the extensive features in jyers

peterADM, 2021-07-07:
> You sure the wiring is correct? Do the colors match the ones in this screenshot?
>
> https://i2.wp.com/smith3d.com/wp-content/uploads/2020/08/7.jpg?w=1000&ssl=1
>
> I read on multiple forums that orange [...]

ill try that wiring, for reference this is the kit i bought 
https://www.amazon.com/gp/product/B08VNYPFCC/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

smooreace, 2021-07-07:
There are at least a dozen Chineesium knock-offs floating around. Not at all good for people hoping for good results

Akujiproxy, 2021-07-07:
> ill try that wiring, for reference this is the kit i bought
> https://www.amazon.com/gp/product/B08VNYPFCC/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

Ok thats not the same wire so don't use my advice above. Instead check the reviews on that amazon page and you will find that someone by the name of "Knappy" has stated that the wire order is indeed incorrect and has shown the correct order in a screenshot there. That should be of some help to you. Get some pliers and fix the wiring order :)

peterADM, 2021-07-07:
> Ok thats not the same wire so don't use my advice above. Instead check the reviews on that amazon page and you will find that someone by the name of "Knappy" has stated that the wire order is indeed [...]

im looking at the review and is he saying that the wiring should be like the picture or?
[image]

Akujiproxy, 2021-07-07:
> im looking at the review and is he saying that the wiring should be like the picture or?

Yeah he's saying its reversed out of the box and it should be like it is in the picture. Mind you another thing it could be is that the magnet is too close, theres a hex key on top of the bltouch, you can create more distance between magnet and needle. Sometimes it pulls too hard to make it pop out and it starts blinking red and doing nothing as well.

peterADM, 2021-07-07:
> Yeah he's saying its reversed out of the box and it should be like it is in the picture. Mind you another thing it could be is that the magnet is too close, theres a hex key on top of the bltouch, [...]

yea my wiring is correct then since its exactly like the picture, ill try the hex key for the magnet

Akujiproxy, 2021-07-07:
> yea my wiring is correct then since its exactly like the picture, ill try the hex key for the magnet

Please do, and oh, did you pull the z-endstop cable out of the mainboard? If not, do that!

peterADM, 2021-07-07:
> > yea my wiring is correct then since its exactly like the picture, ill try the hex key for the magnet
> 
> Please do, and oh, did you pull the z-endstop cable out of the mainboard? If not, do that!

yes a pulled the z-endstop cable and for the hex key magnet, how do i know if its good enough

Akujiproxy, 2021-07-07:
> 
> yes a pulled the z-endstop cable and for the hex key magnet, how do i know if its good enough

I guess i would turn it so far that the needle doesnt go back in instantly when you push it a little bit and then turn it the other way until it does?

peterADM, 2021-07-07:
> > yes a pulled the z-endstop cable and for the hex key magnet, how do i know if its good enough
> 
> I guess i would turn it so far that the needle doesnt go back in instantly when you push it a little bit and then turn it the other way until it does?

Currently running a probe test and the probe worked all 4 times 
i think the issue is resolved. i would never have thought it was the magnet being to close

peterADM, 2021-07-07:
Thanks everyone and Akujiproxy

Akujiproxy, 2021-07-07:
Good to hear and glad to help!

closed 2021-07-07 by peterADM