Jyers/Marlin #1121: Autotilt not functioning as expected

DC-19, 2021-07-07:
### Description

Hi, I recently installed the Jyers Firmware E3V2-UBL-BLTouch-15x15-HS-v4.2.2-v1.3.4, and whenever I try to auto-tilt the bed, it ends up with the probe missing the bed (going too far to the left), then stating that the probing had failed.

### Steps to Reproduce

1. Go into the Leveling Menu
2. Select the auto-tilt the current mesh
**Expected behavior:** [What you expect to happen]
I expect that it would probe the bed again, then compensate for the leveling issues
**Actual behavior:** [What actually happens]
The probe misses the bed, and then states that the probing failed, and returns to the leveling menu
#### Additional Information
I have also tried the 1.3.2 version and the 10x10 versions, to experience the same issues
I am new to this, so I do not know how to find and submit my config files, as my micro-sd card only contains the bin file, even when I save the settings

smooreace, 2021-07-07:
Sounds like you dont have the proper probe offset defined

DC-19, 2021-07-07:
Could you please describe how to adjust this? I am not familiar with the BLtouch, as I only got it a few weeks ago.

Tw0bit, 2021-07-07:
Go to the 3rd box on the Printer Homer Screen, and in that should be area area for defining the Probe Offsets - the distance from the nozzle.

DC-19, 2021-07-07:
It worked. Thanks a million! One more thing, should the probe y-offset be set to 0, since they are level with each other?

Tw0bit, 2021-07-07:
You must measure the distance from the Nozzle to the BLTouch Probe along X (left/right) and Y (forward/back). There are many different mounts. If your Nozzle and Probe are inline, then yes use Y=0.

For example this printable one (https://www.thingiverse.com/thing:4462870) is X=-42 and Y=-10.

DC-19, 2021-07-07:
Sounds Great! Thanks for your help

closed 2021-07-07 by DC-19