Jyers/Marlin #1132: Preheat "Heating Failed"

matthewdolman, 2021-07-10:
### Description

When I preheat, the temps for bed an nozzle do not move, then after about 15 seconds the alarm sounds and I get an error message "Restart Required - Heating Failed".  When I restart the printer the temps for both are clearly much higher than ambient.  It seems as though the temperatures are not being updated after power on.

### Steps to Reproduce

1. Preheat - PLA
2. Wait 15 Seconds
3. Alarm sounds

**Expected behavior:** [What you expect to happen]
Prheat

**Actual behavior:** [What actually happens]
Alarm sounds and preheat not possible.

#### Additional Information

Ender 3 v2 with BLTouch and 4.2.7 board.
[Configuration_adv.zip](https://github.com/Jyers/Marlin/files/6795043/Configuration_adv.zip)

Cal-LF, 2021-07-10:
Tried printing this morning. Preheat PLA. Latest commit compiled successfully.
I get the same heating failed.

Jyers, 2021-07-10:
@Cal-LF Are you able to recreate the temps not moving? If temps are not being reported either there is a major issue in the last marlin pull or there is an individual issue of some kind...

I'll see if I can recreate this myself

Cal-LF, 2021-07-10:
@Jyers Yes, the same thing with both temps. The both stop increasing 79 deg for the nozzle.

darco415, 2021-07-10:
I compiled this morning w/o starting up machine prior. On new flash, heating failed message, lcd panel nor Octoprint reported rise of temperture but both bed and hotend were heating up. I smelled some smoke and shut off machine, checked MB and see the power port with a burnt spot...tried previous flashes, but continues to report heating failure.

Jyers, 2021-07-10:
@darco415 Oh damn, this might be a serious bug from the last pull then. I'm gonna have to revert until we have some more info

darco415, 2021-07-10:
@jyers Ok whew....just re-complied w/your revert....all is NORMAL! Octo and panel reporting temp rise....no more errors

Jyers, 2021-07-10:
@darco415 Is you're board okay? 😅

darco415, 2021-07-10:
@jyers ya...Im running a test print now but looks fine

Jyers, 2021-07-12:
Okay finally got the merge cleaned up

That took... way too long... I'm exhausted...

closed 2021-07-12 by Jyers