Jyers/Marlin #1133: Compile Errors with latest updates

patofoto, 2021-07-10:
### Description

Getting a compile error. Seems that card reader.h is the culprit?.

### Steps to Reproduce

Merged latest update and compiled. 

#### Additional Information

Please forgive me is I am doing this wrong. First time I post an issue. Not too familiar with Git and Github yet.

[image]

[ConfAndTerminal_Files.zip](https://github.com/Jyers/Marlin/files/6795243/ConfAndTerminal_Files.zip)

adam3654, 2021-07-10:
I'm not sure what is causing your issue, but I can tell you that I pulled in the recent commits and successfully compiled about an hour ago.

patofoto, 2021-07-10:
I am getting these errors, but will continue investigating. Assume is because of my firmware configuration.

[image]

adam3654, 2021-07-10:
Part of those errors may be able to be fixed by following the steps [outlined in this post](https://github.com/Jyers/Marlin/discussions/932#discussioncomment-827753)
If I can get to my laptop I'll peek at your configs, but I'm not sure when I'll be able to. #DadLife

patofoto, 2021-07-10:
Thank you @adam3654. Just tried what you suggested and it fixed many of the error messages I was getting except these two:

[image]

adam3654, 2021-07-10:
I just compared what I have for `LONG_FILENAME_HOST_SUPPORT` and in my configuration_adv.h I uncommented `LONG_FILENAME_HOST_SUPPORT` and `SCROLL_LONG_FILENAMES`. Maybe try removing the comment `//` from those lines and try building again?

I'm not sure if that will solve your issue, but it's worth a shot

tititopher68-dev, 2021-07-10:
As @Jyers wrote to me, you have to test all functions and configurations and report any dysfunctions noted. In most cases, it will possibly be the name of variables or the name of function calls to modify, nothing very bad but tedious to identify ...

adam3654, 2021-07-10:
If the compiler is barking about a function not able to be overloaded, it's likely being called with a 'bad' parameter. It wants X and Y is being provided. 

As long as it doesnt break @patofoto 's build, I think uncommenting the `LONG_FILENAME_HOST_SUPPORT` might get him rolling

Jyers, 2021-07-10:
I normally check for compilation error using the templates, so I'll grab your configs and see if I can rectify whatever merge artifact is doing this.

tititopher68-dev, 2021-07-10:
@Jyers 
I open a parenthesis while we are in the fixes.
I think a lot of users don't necessarily use it, I mean the slots for the UBL. In the current code, for the backup you make a G29 S. Only here, if you choose Slot 1 without first activating the UBL, the Create Mesh saves the Mesh on Slot 0. Which is not the goal research...
Too it would be better to replace all the commands where there is a G29 S by:

gcode.process_subcommands_now_P (PSTR ((ubl.storage_slot == 1)? "G29 S1": "G29 S0"));
Because we cannot go beyond 2 slots on the E3V2 ....
or more universally :
char buf [20];
sprintf (buf, "G29 S% i", ubl.storage_slot);
gcode.process_subcommands_now_P (buf);

patofoto, 2021-07-10:
> I just compared what I have for `LONG_FILENAME_HOST_SUPPORT` and in my configuration_adv.h I uncommented `LONG_FILENAME_HOST_SUPPORT` and `SCROLL_LONG_FILENAMES`. Maybe try removing the comment `//` [...]

@adam3654 This worked. It compiled. Not sure why but would be glad continue trying things to help troubleshoot bugs.

@Jyers I am glad to help in any way I can to assist with your branch. I appreciate and enjoy more my printer thanks to you.

tititopher68-dev, 2021-07-10:
@Jyers 
I would add that whatever the choice of slot, it currently always saves in the slot 0 after a Create New Mesh ...
Looking at the Marlin code, when the G29 P0 command is executed, it does a reset (), which forces storage_slot to -1, then when executing the G29 S backup command, Marlin forces storage_slot to 0 .. .

So currently, it is impossible to create a Mesh on slot 1 without deleting slot 0 .... This is not good.

I will delete the G29 P0 command even if it means doing before what this command does except storage_slot = -1
level_state = planner.leveling_active;
set_bed_leveling_enabled (false);
ZERO (mesh_conf.mesh_z_values);

Jyers, 2021-07-10:
@tititopher68-dev Definitely some great observations. After a little more investigation, G29 P1 invalidates the mesh anyway, so no need to even have a zero command there, the G29 P0 can just be removed.

As far as the issue of needing to set the save slot every time, could you elaborate as to what exactly the issue is of needing to activate ubl first. I can't seem to find anything that would indicate activating/deactivating UBL using the `set_bed_leveling_enabled` function would reset the mesh slot.

tititopher68-dev, 2021-07-10:
@Jyers 
In fact, first of all, you should ignore my first post about G29 S1 or S0.  I looked further and found what I wrote in the second post.
 On the other hand, according to Marlin it would be important to correctly initialize the part of the EEprom of the slot before creating the Mesh (this was the goal of the G29 P0).  So I think it is desirable to add these lines (finally move the first 2 and add the third)
 level_state = planner.leveling_active;
 set_bed_leveling_enabled (false);
 ZERO (mesh_conf.mesh_z_values);

Jyers, 2021-07-10:
@tititopher68-dev I think you might be slightly misinterpreting the functioning of the slots as they pertain to the mesh. The way the mesh is stored is really in two places. One mesh is stored directly in RAM (the currently active mesh) while the saved meshes are stored to EEPROM. This means G29 P0 or and other augmentation of the mesh will have no effect on the meshes stored in EEPROM, so zeroing isn't going to initialize anything. All that's really important is that the mesh gets overwritten completely by the new one, and this is achieved when G29 P1 invalidates the current mesh, making all the values undefined.

You can even see from Marlin's own example that G29 P0 is not required for leveling

> G28           ; Home XYZ.
G29 P1        ; Do automated probing of the bed.
G29 P3        ; Smart Fill Repeat until all mesh points are filled in, Used to fill unreachable points.
G29 S0        ; Save UBL mesh points to slot 0 (EEPROM).
G29 F 10.0    ; Set Fade Height for correction at 10.0 mm.
G29 A         ; Activate the UBL System.
M500          ; Save current setup. WARNING - UBL will be active at power up, before any G28.

tititopher68-dev, 2021-07-10:
@Jyers,
Ok, nice

tititopher68-dev, 2021-07-10:
@Jyers 
What do you think of using Marlin's GETTEXT function for menu text, even if it means adding the missing MSGs.  Would that make it possible to have versions in other languages?  And certainly to lighten the code ...
 I also have a version of Alex's plugin for working thumbnails for Cura.  On the other hand, Alex's code seems very dependent on the remaining available Ram and the margin is low ...

Jyers, 2021-07-10:
@tititopher68-dev I've thought a good bit about that and will probably explore it at some point.

Me and Alex are working on caching icons and names for the print screen, hence the use of ram. But we're using the display ram which isn't used for much else, and we have protections in place to ensure it has limits. We've been communicating via discord so far. I don't know if you're familiar with it, but if you want to give it a try I can make a group for us to discuss Aquilla and development ideas. My tag is Jyers#4194

Jyers, 2021-07-12:
Finally got the merge done and dusted.

Git wasn't happy with me... Had to revert the revert from the original merge... Let's just say it's been a long day 😅

closed 2021-07-12 by Jyers

patofoto, 2021-07-12:
Compiled and ready to take for a spin tomorrow. Thank you.