Jyers/Marlin #1136: BL Touch - Fade Mesh level (I think this is the reason) , layer break issue

Vignesh2659, 2021-07-11:
I bought a 3d touch / BL Touch for my Ender 3 V2.  I have tried all the BL touch firmwares for my motherboard V 4.2.2 (except UBL), but I seem to have a layer break issue in between during a print. 

I have attached 2 test prints - a calibration cube and a Benchy. In both of these prints, you can see a white layer which indicates a  break in between. It prints as a whole piece. After removing the object, the print breaks in half (you can see the broken one in the calibration cube), and the same thing is bound to happen in the benchy model.

I have tried this in both the Prusa and Cura slicers. Same issue. There is a fade mesh level option under leveling screen. Initially, it was 10. When I printed it, the break happened at the bottom. Then I increased it to the max (255), hoping it would retain the same mesh for the entire print. But then I observed that the break moved up this time. I feel this might be a reason.

When it was 10, it happened in the bottom, when it is more, it happens a bit up. I have used both the G29 and M420 S1
 gcodes separately, the same problem. Is this issue reported?

Kindly advise how to proceed further.

[image]
[image]

[image]
[image]
[image]
[image]

[image]

Tuncay-Ayhan, 2021-07-11:
Can you share your start code from cura or prusa?

Vignesh2659, 2021-07-11:
Sure. I'll share the Prusa one. Should be the same in Cura as well. This is
the one

*Start G-Code*

M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration

M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate

M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration

M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk

M220 S100 ;Reset Feedrate

M221 S100 ;Reset Flowrate

G21 ; set units to Millimetres

G90 ; absolute positioning

M82 ; set extruder to absolute mode

G28 ;Home

G29 ;BLTouch

G92 E0 ;Reset Extruder

G1 Z2.0 F3000 ;Move Z Axis up

G1 X10.1 Y20 Z50 F5000.0 ;Move to start position and up (nozzle cleaning
possition)

M104 S[first_layer_temperature]

M109 S[first_layer_temperature]

G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position

G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line

G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little

G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line

G92 E0 ;Reset Extruder

*End G-Code*

G91 ;Relative positionning

G1 E-2 F2700 ;Retract a bit

G1 E-2 Z0.2 F2400 ;Retract and raise Z

G1 X5 Y5 F3000 ;Wipe out

G1 Z10 ;Raise Z more

G90 ;Absolute positionning

G1 X0 Y120 ;Present print

M106 S0 ;Turn-off fan

M104 S0 ;Turn-off hotend

M140 S0 ;Turn-off bed

M84 X Y E ;Disable all steppers but Z

I got the above one from internet content. Even before I used this, I used
the default Prusa / Cura slicer G-Code (with the BL touch code enabled),
same problem

Before layer change G-code and After layer G-code are from the default
Ender 3 V2 settings in Prusa. Have not changed anything here

*Before layer change G-code*
;BEFORE_LAYER_CHANGE
G92 E0
;[layer_z]

*After layer G-code*
;AFTER_LAYER_CHANGE
;[layer_z]

Tuncay-Ayhan, 2021-07-11:
If you'd ask me...

The Benchy has a hull line, don't take that as a mistake.
https://help.prusa3d.com/en/article/the-benchy-hull-line_124745/

It also looks like you have z-banding. 

**And just so we know**:

- Was it ok, when using stock firmware? 
- Did you ever printed without those "breaks" ?
- What version Cura / Prusa slicer are you using?
- Does it also occur with the default profiles?

P.S. 

Use UBL, it's nothing special in terms of changes or anything. Just a little bit more longer process time for you mesh being created (depending on which grid you choose). I use it and I think like 90% of the people with BL Touch is using it.

Vignesh2659, 2021-07-11:
Ah okay. But I wonder why is it like that in the Calibration cube as well? The answers to your questions:

- **Was it ok, when using stock firmware?** --> Yes, I just tried the calibration cube once. Have attached the image here. Though the line is seen, it did not break ( I did not want the stock firmware, we all know Jyers' version has everything). Whereas the other messed up an image that is attached here, was the first print I did with the stock Cura / Prusa G-Code settings by using the BLTouch 5*5 bin file. After several z offset settings, I got the perfect one. But then this issue happened.

- **Did you ever printed without those "breaks" ?** --> Before I had BL Touch, I used the Manual Mesh firmware from Jyers, the 5*5 one. Back then everything worked perfectly fine

- **What version Cura / Prusa slicer are you using?****  --> Prusa - V 2.3.0 (based on Slic3r), Cura - V 4.10.0

- **Does it also occur with the default profiles?**** --> If you mean default Cura / Prusa settings? Yeah, same issue. 

Use UBL - Sure. Let me try that as well.

z-banding ? What does this mean, is it the break that happens in between? 

NOTE - The calibration cube I used was the most popular one --> https://www.thingiverse.com/thing:1278865

If you feel there is a better calibration cube or any other quick test prints, please let me know. 

[image]

[image]

[image]
[image]

Tuncay-Ayhan, 2021-07-11:
> But I wonder why is it like that in the Calibration cube as well?

_But I wonder why is it like that in the Calibration cube as well?_
**It can be coincidence**

_z-banding ? What does this mean, is it the break that happens in between?_
This is what I mean: https://www.reddit.com/r/FixMyPrint/comments/hox562/ender_3_v2_banding_explanation_in_comments/

----

What Causes Z Banding in Your Prints?
When a 3D printer user experiences Z banding, it’s usually down to a few main issues:

**In short, these are often the problems** :
- Bad alignment in the Z axis
- Microstepping in stepper motor
- Unstable Z axis rods

I can also see multiple lines that are being weird. Can you share a picture from the left side of your printer. I want to see your leadscrew)  (the long screw). Make a picture as straight as you can from the left side.
[image]

**More questions:**

- How are you excentric nuts? Do you know what that is?
- What layer height are you printing?
- What material are you using?
- What speed are you printing at?

Vignesh2659, 2021-07-11:
Sure. I have attached the images here.

- Excentric nuts --> Yeah I googled, that looks fine, everything is tight
- Layer height --> 1st Layer - 0.35 mm, rest - 0.12 mm
- Material --> PLA
- Speed --> 1st layer - 10 mm/s, rest - 100 mm/s

But when I used the manual mesh 5*5 firmware, these issues did not occur. I did not print the calibration cube back then though. Every other prints worked absolutely fine.

[image]
[image]
[image]

Cal-LF, 2021-07-11:
That looks like a mechanical like Z binding or tight cross bar. I have my fade height set to 2mm and have not seen a skip in the layers like what you have.

Vignesh2659, 2021-07-11:
> That looks like a mechanical like Z binding or tight cross bar. I have my fade height set to 2mm and have not seen a skip in the layers like what you have.

Do you mean the long screw that does the z-axis movement? Or the rod containing the hotend / nozzle set up that moves up and down along with the long screw during z- axis printing?

Also, the mesh fade height, you changed that from the default value of 10 to 2, and haven't seen the skip as it is in my case?

Cal-LF, 2021-07-11:
@Vignesh2659 Yes, either or both would be the first things to check.
I had the fade height set at 10 on a couple prints but never saw skipped layers.

Vignesh2659, 2021-07-11:
> @Vignesh2659 Yes, either or both would be the first things to check.
> I had the fade height set at 10 on a couple prints but never saw skipped layers.

Oh right. It is just that until 2 days back when I used the 5*5 Manual mesh firmware, I never experienced such issues. Let me check though. 

Okay, so the fade height has got nothing to do with it then. Hmm, because when it was at 10, it skipped in the bottom, when I increased it to its highest value (255) , it skipped up. but not below. So I thought this might be a reason.

Vignesh2659, 2021-07-11:
> Oh right. It is just that until 2 days back when I used the 5*5 Manual mesh firmware, I never experienced such issues. Let me check though.
>
> Okay, so the fade height has got nothing to do with it [...]

Vignesh2659, 2021-07-11:
I closed the issue by mistake. My bad, apologies.

Tuncay-Ayhan, 2021-07-11:
> That looks like a mechanical like Z binding or tight cross bar. I have my fade height set to 2mm and have not seen a skip in the layers like what you have.

My guess:

- Excentric nuts on z axis to stiff (or whatever the right word for it is)

@Vignesh2659 Turn your printer off, then pull the plug.  Do the next, very very slow. 
Slowly try to move the x-gantry with 1 hand from bottom to top. Does it move with low or high resistance?

Vignesh2659, 2021-07-11:
There seem to be a high resistance when I move the x gantry from bottom to
top.

However, that resistance is low when I move it from top to bottom

Have tried these with a single hand and a single finger.

Does this mean anything? Are one of the excentric nuts too tight?

Tuncay-Ayhan, 2021-07-11:
> There seem to be a high resistance when I move the x gantry from bottom to
> top.
> 
> However, that resistance is low when I move it from top to bottom
> 
> Have tried these with a single hand and a single finger.
> 
> Does this mean anything? Are one of the excentric nuts too tight?
> 
> 
> > That looks like a mechanical like Z binding or tight cross bar. I have my
> > fade height set to 2mm and have not seen a skip in the layers like what you
> > have.
> >
> > My guess:
> >
> >    - Excentric nuts on z axis to stiff (or whatever the right word for it
> >    is)
> >
> > @Vignesh2659 <https://github.com/Vignesh2659> Turn your printer off, then
> > pull the plug. Do the next, very very slow.
> > Slowly try to move the x-gantry with 1 hand from bottom to top. Does it
> > move with low or high resistance?
> >
> > <https://github.com/Jyers/Marlin/issues/1136#issuecomment-877860015>, or
> > unsubscribe

We are getting somewhere. That means indeed something.

You seem to not have your frame quite square. 

Watch this video. Watch the 
https://youtu.be/JTN6jtB5mqk

From 22 minutes on

Vignesh2659, 2021-07-11:
> #1136 (comment)

Oh alright. Let me check. Its 3 a.m. here. I'll check and update before EOD
today.

Thanks for this. However I still couldn't think how this might have
happened suddenly after getting the 3D touch installed. As I mentioned,
until 2 days back I did not have such an issue.

You can close this thread if needed. I'll reopen it if it is still the
same. I feel that this might be an issue just for me. Cal had confirmed
that it wasn't the case with him.

Tuncay-Ayhan, 2021-07-11:
> Oh alright. Let me check. Its 3 a.m. here. I'll check and update before EOD today. Thanks for this. However I still couldn't think how this might have happened suddenly after getting the 3D touch [...]

Sure thing. It's almost 00.00 for me, so i'll be signing off also. 

I think maybe you messed with the gantry.

Also, while you are at it, install UBL https://github.com/Jyers/Marlin/wiki/Configuration-Details#ubl

**You can use one of these versions:** According to your board in your printer.
4.2.2 version
`E3V2-UBL-BLTouch-10x10-v4.2.2-v1.3.4.bin`

4.2.7 version
`E3V2-UBL-BLTouch-10x10-v4.2.7-v1.3.4.bin`

Also there are HS versions. High Speed. 

If you are sure your bed is leveled "normally", I would go for the HS version. It 's a little faster with probing while creating a mesh. It doesn't retract the probe while moving over the bed. So if you bed is not leveled correctly, It can cause problems.

In probably won't cause issues for you, but I wanted to mention it anyway.

Vignesh2659, 2021-07-12:
Hi, I think the problem is still the same, infact maybe the white layers are more. The Y axis frames were a bit off, that was corrected, also I had put a bit of the grease on the long Z axis screw. I'll send pics soon

Tuncay-Ayhan, 2021-07-12:
> Hi @Tuncay, I think the problem is still there, in fact maybe the white layers are more. The Y-axis frames were a bit off, that was corrected, also I had put a bit of the grease on the long Z-axis screw. The excentric nuts seem to be fine, have not made it too tight or too loose. I'll again do a few prints and check the same with Creality to see if it is still the same

So you got more binding now? So it seems you were having issues with your assembly, rather than with the firmware.

If I were you, I would head over to the Creality discord. Ask some help over there. They can help you with how to redo stuff in a good way. Just head over to one of the # help channels and give them as much as info as you can. 

Good luck buddy. It will print awesome in the end again. Just be patient and update this "issue" when you have fixed your stuff.

Vignesh2659, 2021-07-13:
HI @Tuncay-Ayhan . Thank you so much for the help. Well, I didn't have these issues with the Manual Mash 5 * 5 before I used the 3d touch firmware. But as you suggested, I made things tight, it was still a problem. However, I found a workaround.

I used SUNLU's filament dryer this time, kept the filament dry the entire time it prints. Although not perfect, this made significant differences as attached. So looks like it is not a firmware issue or such. @Cal-LF  had also confirmed. Not sure why, but this looks better compared to before. Hence I am closing this thread with a comment. 

Once again, thank you @Tuncay-Ayhan  for suggesting those videos and your tips, they did make some difference. Good day buddy. Thanks, @Cal-LF. 

[image]
[image]
[image]

closed 2021-07-13 by Vignesh2659

tititopher68-dev, 2021-07-13:
@Vignesh2659 
Your observed print defaults do not seem to come from the firmware. I think for a printer material issue like:
Vref Z-axis Motor,
Screw z-Axis not correctly aligned
Too tight or poorly adjusted Vlsot wheels
Badly adjusted vertical profiles,
etc ...