Jyers/Marlin #1145: Can't resume print after M125 / pause at height

nickbasham, 2021-07-13:
### Description

I'm using Prusaslicer to set an M125 code at a certain layer height, so I can insert magnets into a print. I'm trying to use M125 instead of M0 because it'll park the head and give me room to add the magnets. When it reaches this layer in a print, it pauses, parks the head, displays the M117 message and waits for user input to resume – all good so far. However, when you hit resume, nothing happens. At that point, my only option is to cancel the print. After the confirm cancel dialog, the head returns to the position like it's about to resume, but then displays a filament change dialog. After dismissing that, the job cancels.

I just did a fresh build from the default branch this morning. It's incredibly possible that I'm missing something super basic, but just wanted to flag this in case it is an actual bug.

### Steps to Reproduce

1. Add M125 pause command in sliced file
2. Start print and wait for specified layer

**Expected behavior:**
Clicking resume button resumes the print job after an M125 pause and park command.

**Actual behavior:**
Clicking resume does nothing. When canceling the job, the head returns to the position like it's about to resume, then a filament change dialog is displayed. Canceling the filament change then allows the job cancellation to complete.

#### Additional Information

* Config files, test STL and gcode: 
[M125-Issue-Files.zip](https://github.com/Jyers/Marlin/files/6809858/M125-Issue-Files.zip)
* Video: https://youtu.be/XrNzKOvc_C4

Cal-LF, 2021-07-13:
For what it may be worth, I did a pause at height with Cura and it worked as it should.

nickbasham, 2021-07-13:
@Cal-LF I tested Cura's pause at height post processing just to see how it works, and it uses a different mechanism to achieve the same thing. It uses M0 unconditional stop and then a bunch of relative moves to reposition the head, then moves the head back into the previous position after resuming – it's all calculated based on the start and end positions of the target layers, which is why it requires a script to do those calculations. Theoretically M125 should achieve the same thing with a single command if the firmware supports it. I just prefer PrusaSlicer, but I don't believe it offers any kind of scripting support like Cura does.

Cal-LF, 2021-07-13:
I see. When reading on Marlin gcode, what the M125 command does, it does not do anything after it is parked. Should another gcoded command follow M125?

nickbasham, 2021-07-13:
If you pass the P1 argument, I think it's supposed to prompt the user on the LCD screen and await a response. I was assuming the only valid response is a resume command, and how I was using it, but maybe I misunderstood it.

Jyers, 2021-07-14:
I'll look into this to see if there might be an issue with the resume function

Jyers, 2021-07-14:
@nickbasham

Btw, are you running a precompile bin, or your own compile?

nickbasham, 2021-07-14:
@Jyers I've tested this on the BLTouch 10x10 HS UBL precompile and my own compile using the configs uploaded in the zip file I posted.

I've done some additional digging and looks like others have flagged similar issues in Marlin, so it's doesn't seem limited to this fork:

* https://github.com/MarlinFirmware/Marlin/issues/9254
* https://github.com/MarlinFirmware/Marlin/issues/19870

I think unrelated, but I'm also having an issue with M600 resuming exactly one level too high and causing poor adhesion/failed prints, but I'm having a hard time tracking down whether that's a firmware issue or hardware issue related to z-axis backlash. I'll continue testing/digging and will raise a separate issue if I'm able to track down anything definitive.

Jyers, 2021-07-14:
Hmmm okay it might be todo with how pausing is handled when printing from octoprint... If M125 is called does octoprint go into a paused state, or does it just wait but still think it's printing? Clicking resume when printing from octoprint always sends a host action command telling octoprint to resume, but it octoprint doesn't know how to handle M125 that could be our issue.

nickbasham, 2021-07-14:
Ah, I didn't even think about Octoprint adding another variable to the mix, yep, I bet that's it. I've only been attempting to resume from the printer itself, and didn't go back to check Octoprint until afterwards, but I did see a _Printer Paused / Continue_ dialog. I'll run a test and see if resuming from within Octoprint works.

nickbasham, 2021-07-14:
Alright, I've decided to drop the use of M125 and take advantage of the @pause command code and the after pause and after resume gcode scripts in Octoprint. Resume works properly both in Octoprint and the LCD controller using that setup, so I'll call that a win.

Thanks for the insights @Cal-LF and @Jyers, and keep up the great work, really enjoying this firmware!

closed 2021-07-14 by nickbasham

Jyers, 2021-07-14:
Glad you found a solution. I'll still take a look at the gcode to see if I can improve the handling of M125s when done from octoprint. Shouldn't be too hard. I'll update this if I get it working