Jyers/Marlin #1146: Time to complete is very erratic

befuddled9, 2021-07-13:
**Description**

Time to complete is very erratic. I've seen it go from 8 min to 16 hrs in span of a few minutes.
Using Ender 3 V2 with BLTouch - Firmware version: 1.3.4

**Steps to Reproduce**
Just build any object.

**Expected behavior: [What you expect to happen]**
Actual countdown from build time. Cura reports build time to be ~23hrs

**Actual behavior: [What actually happens]**
Time to complete fluctuates rapidly. Unsure what it is to determine actual duration of build.

https://user-images.githubusercontent.com/5034175/125525236-eff68936-50c0-4190-b13f-5c61b7acf9c9.mp4

Cal-LF, 2021-07-13:
That is the way it is calculating by time/layer. Time remaining gets closer as the print progresses. It is not a bug or firmware failure. Just the way it works.

Jyers, 2021-07-14:
Cal-LF is correct here, the only way the firmware can estimate the time remaining is via linear approximation. Ie if the print is 5% done and it's taken 10 minutes, it will calculate the rate by taking 5/10 and get 1/2 percent per minute, then say if there is 95% left it should take 190 minutes.

If you want accurate times, you can use a post processing script in your slicer to have it insert M73 commands into the gcode, this will display the remaining time based on cura estimates.

closed 2021-07-14 by Jyers