Jyers/Marlin #1151: UBL - Leveling Autotilt of mesh is "broken"

DasSmeagle, 2021-07-15:
### Description

When doing "Autotilt" of (UBL) mesh from display, Leveling is disabled and mesh is destroyed

### Steps to Reproduce

My UBL Leveling is setup for 11 points x and y. I attach my config, it was working in last version a few days ago setup like this, before the "tramming" feature was implemented.

1. Create a mesh (Level -> create new mesh)
Leveling is enabled. "Mesh Viewer" shows the mesh fine. It is saved (can be reloaded, to).
2. Select "Level -> Autotilt current mesh"
3. After Autotilt: 
- Leveling is disabled
- "Mesh Viewer" shows all point same height
4.  Enable Leveling Active [v]
- Mesh Viewer still shows all points of mesh same height
5. Load Mesh
- Mesh Viewer shows the mesh like before step 2

3. [and so on...]

**Expected behavior:** Mesh is tilted and Leveling is active

**Actual behavior:** Mesh is destroyed an Leveling is disabled

[Marlin.zip](https://github.com/Jyers/Marlin/files/6822498/Marlin.zip)

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

DasSmeagle, 2021-07-15:
Update: It also happens when I Autotilt the mesh from gcode while normally printing. Same reaction.

Jyers, 2021-07-15:
If autotilt is completely broken it's an issue from upstream. I'll see if I can recreate it, but if it's like you say there's not much I can do.

DasSmeagle, 2021-07-15:
OK, Thanks.

My config did work before, I double checked it for errors.

DasSmeagle, 2021-07-16:
Yes, found it. 

https://github.com/MarlinFirmware/Marlin/issues/22175

Broken for 30 days now :(

Maybe stick to the released versions for now, "they" broke a lot when adding the new axes... 

Guess I have to go to an older release now, I hate it.

Jyers, 2021-07-17:
I've created a new branch for UBL users until Marlin fixes this... https://github.com/Jyers/Marlin/tree/E3V2-Temp-Revert-for-UBL

DasSmeagle, 2021-07-18:
@Jyers GREAT, Thank you very much!!!

bwspath, 2021-07-19:
https://github.com/MarlinFirmware/Marlin/pull/22398 should fix this too

Jyers, 2021-07-19:
@bwspath Well done! Thanks you 😄

DerAndere1, 2021-07-19:
This issue was indeed introduced with the addition of support for more axes. Addition of  these axes was the product of the joint effort of 8 contributors, including me and thinkyhead. We did our best to test things but the final merge was still challenging (5000 lines of code in 100 files) and the bed leveling bug sneaked in during final clean-up. It is great to see the regression in bed leveling fixed with pull request 22398 and we wish everyone a joyful experience with Marlin, once the PR by bwspath is merged.

Jyers, 2021-07-20:
Thanks @DerAndere1 I'll pull the fix, you all are doing great work upstream!

closed 2021-07-20 by Jyers

DasSmeagle, 2021-07-26:
@Jyers 

Yesterday I had to print something, so I tested the temporary "E3V2-Temp-Revert-for-UBL" branch.

Sorry, it doesn't work 100% :( basic functionality it there, UBL working, so I could print my project. But manual Bed Leveling and Assisted Tramming is broken, head won't move when changing corners.

But no need to fix that, as UBL seems to be fixed in Marlin just delete that temporary branch and pull the fixes to the main branch. If I see it right, that has not happen, yet, right?

Thanks,
- Oliver