Jyers/Marlin #1152: Heater failed error during filament change after heater timeout

Nepoxx, 2021-07-15:
### Description

Executing a `M600` command mid-print to change filament results in the heater timing out after some time (which is fine and great), but then upon restarting the heater, the printer will eventually panic and complain about a failed heater, requiring restart and thus failing the print (resuming does not work because of the displacement caused by the filament change not being accounted for). This occurs even though the temperature was rising normally and occurs "far way" from the target temperature.

### Steps to Reproduce

1. Add `M600` to a print (or send it from terminal)
2. Wait for heater to time out and temperature to drop
3. Restart heater from LCD's prompt
4. Temperature rises normally but printer panics after some time, way short of target temperature

#### Additional Information

Might be related to https://github.com/Jyers/Marlin/issues/1065

-----
Increasing the thermal protection timeout does not resolve the issue. I am not able to reproduce 100% of the time, but it occurs msot of the time (9 times out of 10), probably has something to do with timing. My uneducated guess is that the thermal runaway protection does not consider the temperature drop due to heater timeout, and thus assumes failure because the temperature is not matching the target temperature.

Nepoxx, 2021-07-15:
Upon further research, I cam upon this issue on the upstream repo: https://github.com/MarlinFirmware/Marlin/issues/20616
So this is most likely an issue with Marlin itself. Should I close this isssue and re-open it over there?

Cal-LF, 2021-07-15:
Does the printer have the latest version of firmware?

Nepoxx, 2021-07-15:
Yes, v1.3.4 (sorry should have speicifed)

Nepoxx, 2021-07-26:
If I let the hotend cool down completely before restarting heating, I don't have the issue. This is definitely the same issue as https://github.com/Jyers/Marlin/issues/1065

closed 2021-07-26 by Nepoxx